genodelabs/genode #3093: RISCV simulation on spike hangs

johnnavaj, 2018-12-20:
Apart from having issues with linking , only make run/log works but hangs on spike with a boot time out.

I am guessing its something to do with the bbl image ? 

using 'ld-hw.lib.so' as 'ld.lib.so'
core link address is 0xffffffc000000000
using 'var/run/log/boot/image.elf' as bbl image
/home/john/John/Projects/genode/18-hw/genode/contrib/bbl-30243a733d49c8545adbb230df7e01f8a8fc12a0/src/lib/bbl/bbl/mtrap.c: In function ‘mcall_trap’:
/home/john/John/Projects/genode/18-hw/genode/contrib/bbl-30243a733d49c8545adbb230df7e01f8a8fc12a0/src/lib/bbl/bbl/mtrap.c:289:12: warning: ‘retval’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   regs[10] = retval;
   ~~~~~~~~~^~~~~~~~
spawn spike -m128 /home/john/John/Projects/genode/18-hw/genode/tool/riscv_v2/debug/bbl
/Boot process timed out
error writing "stdout": bad file number
    while executing
"puts "Log step failed, retry.""
    (procedure "run_genode_until" line 34)
    invoked from within
"run_genode_until "Test done.*\n" 20"
    (file "/home/john/John/Projects/genode/18-hw/genode/repos/base/run/log.run" line 28)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/home/john/John/Projects/genode/18-hw/genode/tool/run/run" line 971)
Makefile:319: recipe for target 'run/log' failed
make: *** [run/log] Error 1

chelmuth, 2018-12-20:
For me it works on Genode 18.11 with spike 17.05 (from https://github.com/ssumpf/riscv-isa-sim.git)

```
core link address is 0xffffffc000000000
using 'var/run/log/boot/image.elf' as bbl image
/home/krishna/src/genode.git/contrib/bbl-30243a733d49c8545adbb230df7e01f8a8fc12a0/src/lib/bbl/bbl/mtrap.c: In function ‘mcall_trap’:
/home/krishna/src/genode.git/contrib/bbl-30243a733d49c8545adbb230df7e01f8a8fc12a0/src/lib/bbl/bbl/mtrap.c:289:12: warning: ‘retval’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   regs[10] = retval;
   ~~~~~~~~~^~~~~~~~
spawn spike -m128 /home/krishna/src/genode.git/build/riscv_spike/debug/bbl
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv
rr                vvvvvvvvvvvvvvvvvvvvvv
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
starting bootstrap at 0000000081000000 ...

kernel initialized
ROM modules:
 ROM: [000000008114d000,000000008114d156) config
 ROM: [00000000812c6000,00000000812c7000) core_log
 ROM: [0000000081107000,0000000081145d20) init
 ROM: [000000008114e000,00000000811f02a8) ld.lib.so
 ROM: [0000000081146000,000000008114c6a0) test-log

Genode 18.11
106 MiB RAM and 64534 caps assigned to init
[init -> test-log] hex range:          [0e00,1680)
[init -> test-log] empty hex range:    [0abc0000,0abc0000) (empty!)
[init -> test-log] hex range to limit: [f8,ff]
[init -> test-log] invalid hex range:  [f8,08) (overflow!)
[init -> test-log] negative hex char:  0xfe
[init -> test-log] positive hex char:  0x02
[init -> test-log] multiarg string:    "parent -> child.7"
[init -> test-log] String(Hex(3)):     0x3
[init -> test-log] Very long messages:
[init -> test-log -> log] 1.....................................................................................................................................................................................................................................2
[init -> test-log] 3.....................................................................................................................................................................................................................................4
[init -> test-log] 5.....................................................................................................................................................................................................................................6
[init -> test-log]
[init -> test-log] Test done.

Run script execution successful.
```

Please try to start from fresh build directory (`./tool/create_builddir riscv_spike BUILD_DIR=fresh`) with the spike version mentioned above.

johnnavaj, 2018-12-20:
That is weird as you are using a different version of Spike along with the fevsr also .  I tried compiling the fesvr and spike as provided by ssumpf(17.05) , it fails in spike for the gdbserver.cc. So i am not sure what exactly is wrong with the bbl and spike or it is something else ?

johnnavaj, 2018-12-21:
@chelmuth with ssumpf's repo of the fesvr and spike its working . but any idea why it hangs on spike repo of the riscv developers ?

chelmuth, 2018-12-21:
Let's wait what @ssumpf has to tell us.

ssumpf, 2018-12-21:
@johnnavaj: Because the developer repo implements a different privileged ISA. We have support for version 1.9.1. The current version is 1.10 and spike will most likely implement some experimental version. This is why I had to fork these repos.

ssumpf, 2018-12-21:
With the next GCC update we will have to update to 1.10 or later, have upstream GCC support for free, and may switch from spike to Qemu, which now supports RISC-V properly. From my side and if everyone feels comfortable we may close this issue.

johnnavaj, 2018-12-21:
@ssumpf thanks for the response . Will work with the current one as provided until a new patch comes out.

closed 2018-12-21 by johnnavaj