genodelabs/genode #3098: Menu view: custom font color schemes

ehmry, 2019-01-03:
I've been working on a high-contrast menu view style and would like to use a light colored font against a dark background. Perhaps A light and dark font color could be specified in a configuration file included with the PNGs and the font with the greatest contrast can be selected automatically for a given background, or fonts for each background could be specified by configuration.

ehmry, 2019-01-11:
I have an initial implementation for setting font color in additional style metadata, but perhaps widget margins should be overrideable as well...

alex-ab, 2019-01-11:
Just to acknowledge the demand, @cnuke and me also have already a bunch of [ad-hoc changes to menu_view](https://github.com/alex-ab/genode/commits/sculpt_pkg_top_view_18_11) just to operate top_view.

ehmry, 2019-01-11:
I agree that simply adding a text palette is too ad-hoc. Another other thing to consider is scaling margins on high resolution monitors.

nfeske, 2019-01-11:
@ehmry the benefit of using the VFS for supplying the style information is that high-DPI support can be provided from the outside of the menu view by a component that updates the style depending on the resolution. Sculpt already uses this approach to great success for the font_fs, which is configured by the sculpt manager depending on the screen size. It magically makes the terminal high-DPI ready without the terminal's support. By following this approach for the menu view's styling, the menu view implementation remains unbothered with the complexities of abstracting from physical pixels.

nfeske, 2020-02-28:
With Genode 20.02, the color of labels can be styled.

closed 2020-02-28 by nfeske