genodelabs/genode #3101: Move timer/timeout facilities from os to base repository

nfeske, 2019-01-03:
The timer drivers as well as the timeout-handling libraries should move to the base repository because the code is included in the dynamic linker (hosted in base) and the drivers are included in the various base-<kernel> depot archives. By moving the timer handling code to base, the source-tree structure becomes more logical and several special cases for assembling depot content can be removed.

nfeske, 2019-01-08:
In the squash commit above, I took the opportunity to move the timer from src/drivers/timer to src/timer. Since the timer is now hosted in base, the drivers directory won't ever contain any other driver anyway. One also could argue whether or not the timer is driver. On some kernels, it is. On others it isn't. Moreover, its name is traditionally not suffixed with `_drv` (in contrast to all other drivers). So by moving it directly to src, we acknowledge that the timer is something special.

nfeske, 2019-01-14:
Merged to master.

closed 2019-01-14 by nfeske

cproc, 2019-05-07:
While doing compile tests with disabled os repository I noticed that two 'time_source.h' files include <os/attached_mmio.h>. Should this header be moved to base as well?

nfeske, 2019-05-07:
I think that this header is a bit artificial. It's fine to keep it but I'd rather not promote it to become part of the base repository. Would it be possible to adapt the time_source.h instances to not use it?

chelmuth, 2019-05-08:
IMO the private inheritance strongly suggests the `Attached_io_mem` should be aggregated and combined with `Mmio` explicitly.

m-stein, 2019-05-08:
Attached_mmio reflects the most common use pattern of Mmio which is used in more than 30 places. It's not that important for me if you remove it, but I'm a bit confused by the rational.

nfeske, 2019-05-08:
@m-stein please re-read my comment. I have not proposed to remove it.

m-stein, 2019-05-08:
Sorry, I'll re-phrase my comment: Could you please explain what you mean with "is a bit artificial"?

nfeske, 2019-05-08:
My comment was not meant to devalue your work. Sorry if it came across as smug. I just meant that the class is a trivial wrapper after all. Please re-read the documentation of the class (https://github.com/genodelabs/genode/blob/master/repos/os/include/os/attached_mmio.h#L24). Then read the code. Then you will probably see why the term "artificial" crossed my mind. ;-)

I agree that the utility is handy and avoids repetition. But it is not fundamental to make Genode work, nor it acts as a safeguard against common bugs, like the other code in the base repository. It exists solely for convenience. For this reason, it should better stay outside the base repository. To illustrate my point by counting beans: `Attached_mmio` is larger than the repeated amount of code within base that it would avoid. For the os repository and above, the stats are far more positive. So the os repository is a natural place.

While there is no tangible benefit for having the header in base, the move of the header from os to base would require us to adjust at least 28 places + explain the change to 3rd-party developers. That is not justifiable in my opinion.

m-stein, 2019-05-08:
Thanks for outlining your thoughts. Sorry for sounding a bit harsh this morning, I just wanted to understand the decision.