genodelabs/genode #3106: Enable optimization level O3 by default

nfeske, 2019-01-07:
In https://github.com/genodelabs/genode/blob/master/repos/base/mk/global.mk#L135 the Genode build system defines the use of O2 as default optimization level. Because of this conservative choice, Genode currently does not benefit from many compiler optimizations (e.g., the memcpy throughput with O3 is significantly better). I propose increasing the level to O3 by default. Components that run into problems with O3 should either be fixed or explicitly compiled with O2 by setting `CC_OLEVEL := O2` in their respective target.mk files.

nfeske, 2019-01-07:
BTW, I already gave this a quick try. Compiling the code base seems to work fine at most. I only encountered a problem when building the server/rom_filter component. There seems to be a problem with the use of the default assignment operator for the `String` class, which is probably revealed by O3's more thorough information-flow analysis. I haven't further investigated the issue, but I assume that the explicit implementation of `operator =` could solve it.

That said, the elephant in the room is of course how the compiled binaries are behaving.

ttcoder, 2019-01-07:
> There seems to be a problem with the use of the default assignment operator for the `String` class, which is probably revealed by O3's more thorough information-flow analysis. I haven't further investigated the issue, but I assume that the explicit implementation of `operator =` could solve it.
> 

I might be doing a *completely* irrelevant observation here, and if so disregard it; but when I initially tried to compile Genode last november, it was with GCC 7.3 (long story :-)

And it mentionned a few dozen warnings/errors; one of the things that prevented from compiling Genode was, IIRC, with a String<150..> template versus a Session_label (template?); for some reason it only proceeded with compiling if I shortened the first templates to 140 characters so that it would "fit" the session label.

Anyway if that happens to be relevant, I guess I'm saying it could be interesting to compile with gcc7.3 (not necessarily one that is configured fully as a cross compiler with regard to correct TLS configuration ..etc: the goal would be to pinpoint error lines, then to return to the normal 6.3 toolchain, NOT to use 7.3 as the end-to-end toolchain).

nfeske, 2021-02-26:
There is no current plan to follow up on this issue as global policy. Let's instead try to selectively build individual components with O3.

closed 2021-02-26 by nfeske