genodelabs/genode #3108: foc: stack_area is reserved wrongly

alex-ab, 2019-01-07:
Components on Genode/FOC use [this generic stack area base]( https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/base/stack_area_addr.cc#L17), however core on FOC uses [another stack area base](https://github.com/genodelabs/genode/blob/master/repos/base-foc/src/core/stack_area_addr.cc#L24). The modified stack area base is used to setup the [utcb area for FOC threads](https://github.com/genodelabs/genode/blob/master/repos/base-foc/src/core/include/platform_pd.h#L60) in the [components address space](https://github.com/genodelabs/genode/blob/master/repos/base-foc/src/core/platform_pd.cc#L139). Unfortunately the values differs, so that the stack area of the components is not reserved and can be used for whatever attachments, which leads to corrupted UTCB states.

alex-ab, 2019-01-07:
The commit avoids the issue and fixes my corruption I have seen, however I'm not sure whether it is the best way ...

nfeske, 2019-01-08:
Your patch definitely makes sense. Thanks Alex! Merged to staging.

nfeske, 2019-01-09:
Thanks to an offline discussion this morning, I realized that I misinterpreted @alex-ab's patch. To recapture the problem:

* We need to reserve a virtual-address area in each component where UTCBs are located (let's call it utcb area). This area must not be used for attaching any dataspaces. For this reason, we use to place the utcb area within the stack area, which is a reserved virtual-address area already.
* The base address of the stack area differs between core and other components because core's virtual address space layout is constrained in several ways. I.e., on Fiasco.OC a piece of physical memory can only be used if it is mapped in core's virtual address space because a mapping operation (targeting any address space) takes a core-local address as mapping-source argument. For this reason, core's has a one-to-one mapping of the physical to its virtual address space. So we have to place the stack area somewhere where no physical memory and MMIO is located. Unfortunately, the physical memory layout differs a lot between different platforms (e.g., on PCs, RAM usually starts at 0, but on some ARM devices, physical RAM starts at a rather high address). Hence, we tried to pick an address for core's stack area that is free from conflicts on all supported devices. We found that 0x20000000UL worked quite well.
* The bottom line of the previous point is that core's stack area is located at a different address than other non-core components, for valid reasons. The problem that Alex observed is the part where core sets up the utcb area for a non-core component. It tries to place the utcb area at a certain position within the stack area. The problem is that it uses core's stack area as the basis for this computation, not the stack-area address for regular components.
* Core does not know the stack area base address of regular components because this information is only available in the base library (the one linked to regular components).

To solve this issue in a clean way, we should declare the non-core stack-area base address at some place that is accessible by both core and the base library. I suggest extending the interface of base/src/internal/stack_area.h with a new function `stack_area_virtual_base_non_core`, which returns the regular base address when called within core. There exists only one implementation of this function, for core on the Fiasco.OC base platform. The actual address can then be defined at a new header at base-foc/src/include/base/internal/stack_area_addr.h, which is included by both the stack_area_addr.cc of core and stack_area_addr.cc of the base library.

nfeske, 2019-01-09:
44de8d979a631dc1ca2073355aec279686015975 is my take on the issue. @alex-ab would you like to give it a try?

nfeske, 2019-01-09:
BTW, please note that I reverted your fix from staging.

alex-ab, 2019-01-09:
Thanks to recap our offline discussion. Your commit seems however to be somehow incomplete, since the stack addr value is still defined two times, in repos/base/src/lib/base/stack_area_addr.cc and repos/base-foc/src/include/base/internal/non_core_stack_area_addr.h. Whenever one of the values changes the next time, the error comes bug silently and unnoticed as it did this time. I would propose not to use two values, but instead one. Please consider bc19f02a40c3da330f35f940f6327ef8a2133d1e  as fixup

nfeske, 2019-01-09:
With my commit, base-foc no longer uses `base/src/lib/base/stack_area_addr.cc` but instead it uses the new file `base-foc/src/lib/base/stack_area_addr.cc`. The latter includes the `non_core_stack_area_addr.h`, which is also local to base-foc. I briefly considered the approach you took with your fixup but feared that this might be confusing for users of base-nova. Here, the stack area base is at 0xa0000000UL (https://github.com/genodelabs/genode/blob/master/repos/base-nova/src/lib/base/stack_area_addr.cc). So looking at the generic header may be misleading.

That said, your criticism is valid because the accidental linking of the base version of `stack_area_addr.cc` instead of the base-foc version would go unnoticed. Renaming the base-foc version to `stack_area_addr_foc.cc` would help here. What do you think?

alex-ab, 2019-01-09:
I think the special case for base-nova will/can be dropped when the VM experiments hit master, so the misleading argument would not apply anymore. But up to this point, I see the argument. I'm fine with whatever way you choose.

nfeske, 2019-01-09:
> I think the special case for base-nova will/can be dropped when the VM experiments hit master, so the misleading argument would not apply anymore.

That's all the better. So I'll cherry-pick your fixup.

commit b8645187 (2019-01-08): foc: use common stack area base for core and apps

Fixes #3108
commit 44de8d97 (2019-01-09): base-foc: fix placement of utcb area stack area

This commit ensures that UTCB areas of PDs are positioned relative to
the stack areas of regular components, not the one of core.

Fixes #3108
commit 97e3d05f (2019-01-14): base-foc: fix placement of utcb area stack area

This commit ensures that UTCB areas of PDs are positioned relative to
the stack areas of regular components, not the one of core.

Fixes #3108
closed 2019-01-14 by nfeske