genodelabs/genode #3109: Possible deadlock on signal context dissolving

cproc, 2019-01-08:
When the signal receiver thread receives a signal, it calls ```Signal_context_registry::test_and_lock()```, which acquires the registry lock, looks up the signal context and tries to lock the context. If the context is already locked by another thread calling ```Signal_receiver::dissolve()```, a deadlock occurs when this thread calls ```Signal_context_registry::remove()``` and the registry is already locked by the signal receiver thread.

With the added debug messages of commit [1], the problem is quite reproducible with the 'noux_tool_chain_auto' test on a Lenovo x260.

[1] https://github.com/cproc/genode/commit/16ad7f957adf16afac3f6c3d4769a52af4d9416a

nfeske, 2019-01-29:
The commit above solves this issue for the test case I have (test-timed_semaphore on NOVA/x86_32/Qemu). @cproc Could you give it a try in your test scenario?

nfeske, 2019-01-29:
I merged the fix to staging after reviewing it together with @chelmuth. I'd still greatly appreciate @cproc testing it with the noux scenario.

cproc, 2019-01-29:
I could not reproduce the deadlock with the fix commit so far.

commit 07a13d81 (2019-01-29): base: fix deadlock during signal-context dissolve

This patch moves the removal of the signal context from the
'_platform_finish_dissolve' to the '_platform_begin_dissolve'
method. This is needed because the removal involves taking
the signal-registry lock. The latter must adhere the same
locking order as the code path used for signal delivery.

Fixes #3109
closed 2019-01-30 by nfeske