genodelabs/genode #3115: qt5: support headers and resources in other directories in qmake project files

cproc, 2019-01-11:
Support paths like '../util.h' or 'resources/panel.qrc' in the HEADERS and RESOURCES variables in qmake project files.

chelmuth, 2019-01-15:
Unfortunately, last nights package build failed with the following errors and I assume this is due to the implemented approach inherently not supporting parallel builds. All independent *tiled_wm* targets now depend on *../moc_util.o* and may create the file in parallel.

```
  Program test/tiled_wm/manager/test-tiled_wm-manager
    COMPILE  main.o
    ASSEMBLE qt5_core.symbols.o
    ASSEMBLE qt5_gui.symbols.o
    MERGE    qt5_core.abi.so
    ASSEMBLE qt5_widgets.symbols.o
    MERGE    qt5_gui.abi.so
    MERGE    qt5_widgets.abi.so
  Program test/tiled_wm/panel/test-tiled_wm-panel
    CONVERT  ../moc_util.cpp
    CONVERT  moc_panel.cpp
    CONVERT  qrc_panel.cpp
  Program test/tiled_wm/app/test-tiled_wm-app
  Program test/tiled_wm/overlay/test-tiled_wm-overlay
    CONVERT  ../moc_util.cpp
    CONVERT  moc_app.cpp
    CONVERT  moc_icon.cpp
    CONVERT  ../moc_util.cpp
    CONVERT  moc_overlay.cpp
    COMPILE  main.o
    CONVERT  qrc_app.cpp
    COMPILE  panel.o
    CONVERT  qrc_overlay.cpp
    COMPILE  app.o
    COMPILE  qrc_panel.o
    COMPILE  main.o
    COMPILE  main.o
    COMPILE  overlay.o
    COMPILE  qrc_app.o
    COMPILE  qrc_overlay.o
    COMPILE  ../moc_util.o
    COMPILE  moc_icon.o
    COMPILE  moc_panel.o
    COMPILE  moc_app.o
    COMPILE  moc_overlay.o
    COMPILE  ../moc_util.o
    COMPILE  ../moc_util.o
    LINK     test-tiled_wm-overlay
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: ../moc_util.o: TLS transition from R_386_TLS_IE to R_386_TLS_LE_32 against `' at 0x5102 in section `.text._ZNK19Genode_signal_proxy10metaObjectEv' failed
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
/data/genode/repos/base/mk/prg.mk:193: recipe for target 'test-tiled_wm-overlay' failed
make[6]: *** [test-tiled_wm-overlay] Error 1
make[6]: Target 'all' not remade because of errors.
var/libdeps:265: recipe for target 'test-tiled_wm-overlay.prg' failed
make[5]: *** [test-tiled_wm-overlay.prg] Error 2
    LINK     test-tiled_wm-manager
    LINK     test-tiled_wm-app
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: ../moc_util.o: TLS transition from R_386_TLS_IE to R_386_TLS_LE_32 against `' at 0x5102 in section `.text._ZNK19Genode_signal_proxy10metaObjectEv' failed
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
/data/genode/repos/base/mk/prg.mk:193: recipe for target 'test-tiled_wm-app' failed
make[6]: *** [test-tiled_wm-app] Error 1
make[6]: Target 'all' not remade because of errors.
var/libdeps:229: recipe for target 'test-tiled_wm-app.prg' failed
make[5]: *** [test-tiled_wm-app.prg] Error 2
    LINK     test-tiled_wm-panel
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: ../moc_util.o: TLS transition from R_386_TLS_IE to R_386_TLS_LE_32 against `' at 0x5102 in section `.text._ZNK19Genode_signal_proxy10metaObjectEv' failed
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: error in ../moc_util.o(.eh_frame); no .eh_frame_hdr table will be created.
/usr/local/genode-gcc-1705/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
/data/genode/repos/base/mk/prg.mk:193: recipe for target 'test-tiled_wm-panel' failed
make[6]: *** [test-tiled_wm-panel] Error 1
```

cproc, 2019-01-16:
Commit edc8ad2 should fix the parallel build problem. I moved the parts related to qmake project files out of the qt5 import file, because the qmake project files are currently only supported for applications and the added vpath statements use the 'PRG_DIR' variable.

nfeske, 2019-02-12:
Merged into master.

commit 119a12cb (2019-01-14): qt5: support relative paths in qmake project files

Support paths like '../util.h' or 'resources/panel.qrc' in the HEADERS and
RESOURCES variables in qmake project files.

Fixes #3115
closed 2019-02-12 by nfeske