genodelabs/genode #3116: Documentation of Socket FS

chelmuth, 2019-01-14:
With #3113, another feature (support for non-blocking connect) makes it into the socket fs, which still lacks a documentation of its structure and semantics. This issue is a collection point for all information of the current state of supported socket semantics, files in the fs, and maybe known shortcomings.

ehmry, 2019-01-16:
I have some initial documenation at df22ccbec2db21a5cb441293d41a0ff1a4d31096. Could use cleaner formating and TCP examples.

chelmuth, 2019-03-19:
As @ehmry answered to my commit comment the initial documentation is just a reference manual for the directory structure and the file semantics. We have to add a section about exemplary use of the file system from network client and server perspective, which

- can leave out some details in a simple example
- may provide more complex example(s) for multi-connection servers/clients
- can be extended example(s) for blocking vs. non-blocking

nfeske, 2021-02-26:
This issue has become stale.

closed 2021-02-26 by nfeske