genodelabs/genode #3120: run/depot_autopilot: less interactive mode

m-stein, 2019-01-17:
This should avoid that one missing package is causing all test of a platform to fail:
bc2f8b0e48 run/depot_autopilot: less interactive mode
51fe1137fd run/depot_autopilot: check all archives at once

nfeske, 2019-01-18:
@m-stein Can both commits merged into one or do you deem their separation as important?

After yesterday's chat with @senier about the work flow of integrating and running tests manually, how do you think about condensing this work-flow practice into a posting for genodians.org? This would be very valuable for others too (including me).

m-stein, 2019-01-18:
No problem, you can squash them together. Sure, I try to write some lines.

nfeske, 2019-01-18:
Thanks. I merged all three commits squashed together as https://github.com/genodelabs/genode/commit/dfcd4142a4a86c867ee513e808bbe9e7e0d54fce to staging.

commit dfcd4142 (2019-01-18): run/depot_autopilot: less interactive mode

In less interactive mode, the run script doesn't give up on missing test
archives but instead removes the corresponding tests and marks them "missing".
This mode avoids total failure of a platform in automated test infrastructures
when only a few archives are missing.

Fixes #3120
commit ffd4e231 (2019-01-30): run/depot_autopilot: less interactive mode

In less interactive mode, the run script doesn't give up on missing test
archives but instead removes the corresponding tests and marks them "missing".
This mode avoids total failure of a platform in automated test infrastructures
when only a few archives are missing.

Fixes #3120
closed 2019-01-30 by nfeske