genodelabs/genode #3122: netperf: remove timer.cc (implementation of `start_timer`, 'times_up')

nfeske, 2019-01-18:
The port of netperf uses an ad-hoc implementation of a `start_timer` interface, which relies on deprecated Genode API calls (to create a thread) and the alarm library. The latter I want to remove.

@alex-ab Can you remember why this implementation was needed in the first place? Do you see a way to dispose this implementation or to implement it using plain libc primitives?

alex-ab, 2019-01-18:
I have no idea. A quick look seems that the [times_up variable](https://github.com/genodelabs/genode/blob/master/repos/ports/src/app/netperf/timer.cc#L53) must be set on a regular interval. I'm not firm with the todays libc primitives and this integrates well with netperf, sorry.

chelmuth, 2019-01-23:
The netperf implementation depends on an alarm timer for user defined test timeouts. On UNIX and BSD, the original `start_timer` implementation uses `alarm()` and a signal handler for `SIGALRM`, which also determines the timer semantics. A stop-gap solution could implement `start_timer()` via a pthread that calls `sleep()` but the final solution would be support for `alarm()`.

chelmuth, 2019-02-07:
Please find the pthread-based implementation in dadcfa3a3e68fd31849e00ac727f98a2f3c55b71.

nfeske, 2019-02-12:
Fixed in master.

closed 2019-02-12 by nfeske