genodelabs/genode #3126: Add getentropy to libc

ehmry, 2019-01-21:
I have always been uncomfortable with using `/dev/random` on Genode because I worry that if this file is missing an application may fall back to the using a time or fixed value as a random seed. I propose to preemptively add the `getentropy` call to our libc and patch applications to remove the use of `/dev/random` and `/dev/urandom` (I don't know the difference between the two, and don't care, having two is just bad).

`getentropy` is available in [OpenBSD](https://man.openbsd.org/getentropy), [Glibc](https://www.gnu.org/software/libc/manual/html_node/Unpredictable-Bytes.html#index-getentropy), and is a proposed addition to [POSIX](http://austingroupbugs.net/view.php?id=1134).

I'm not sure what the best backend for `getentropy` would be, but at least we would be able to detect which applications are using it. Perhaps we could add another configuration to the libc in the same manner as stdio or Rtc and configure the backend as a file, such as `<libc ... entropy="/dev/entropy"/>`. If no backend is explicitly configured, the application is killed.

chelmuth, 2019-03-04:
I'm feeling torn between the definite approval of this quite reasonable proposal and the fear of the unknown issues in practical applicability. Did you already analyze the entirety of our ports? Also, do you think this task should experience priority? I'm afraid if it's not addressed soon with the current state of documentation and implementation progress it may just be left to rot here.

ehmry, 2019-03-04:
Should the libc be updated, this would be an implicit feature, so I will close this and we can deal with it when it comes.

closed 2019-03-04 by ehmry