genodelabs/genode #3131: Update libpng to 1.6.36

ehmry, 2019-01-24:


ehmry, 2019-01-24:
Fix at e961c3268713daa59fd0bd7e86be400c2b43096d.

nfeske, 2019-01-24:
This commit breaks the build of the menu_view binary archive:

```
In file included from .../api/gems/2019-01-24/include/gems/png_image.h:21:0,
                 from .../src/menu_view/2019-01-24/src/app/menu_view/style_database.h:19,
                 from .../src/menu_view/2019-01-24/src/app/menu_view/widget_factory.h:18,
                 from .../src/menu_view/2019-01-24/src/app/menu_view/main.cc:15:
.../api/libpng/2019-01-24/include/png.h:330:27: fatal error: pnglibconf.h: No such file or directory
 #   include "pnglibconf.h"
```

ehmry, 2019-01-25:
Second attempt: 0a20477d0144b38e4323f6d5531d3634633ed049.

nfeske, 2019-01-30:
I merged the second attempt to staging now.

nfeske, 2019-01-31:
The build fails on in rpi:

```

build/hw_rpi $ make LIB=libpng
...
In file included from /.../contrib/libpng-ac05f35990279bbd8cc2f019ca51bc5d17b32be6/src/lib/libpng/pngerror.c:19:0:
/.../contrib/libpng-ac05f35990279bbd8cc2f019ca51bc5d17b32be6/src/lib/libpng/pngpriv.h:72:26: fatal error: pnglibconf.h: No such file or directory
 #  include "pnglibconf.h"
....

nfeske, 2019-01-31:
Hmmm... the same happens on x86_64. May I have picked the wrong commit?

ehmry, 2019-01-31:
I have two fixups, 73af3284eda4534c77f39e8ad3fe7541a7678b8f and 8859877dc11bcbc30928e2cd2bf159e1f8dd62fb. Tested with `run/mudpdf`.

nfeske, 2019-01-31:
Looks good with your fixups. Also building the pdf_view pkg works well.

ehmry, 2019-02-06:
I have a patch 243b271d23ef01e02dea8dd755ad7513a870b749 to ~bloat~ add more symbols to libpng, this hopefully adds the symbols that `qt5_webcore` is looking for.

nfeske, 2019-02-06:
Thank you @ehmry. I merged your fixup to staging.

ehmry, 2019-02-06:
I have a fixup to the last fixup at fdf2458c7563ea5169b9be1c543858947b3c3306, @cnuke pointed out that upstream publish a list of symbols.

nfeske, 2019-02-06:
I merged this one also. ;-)

commit 74382a0b (2019-01-24): Update libpng to version 1.6.36

Fix #3131
commit 89e2064d (2019-01-30): Update libpng to version 1.6.36

Fix #3131
commit 13268ec4 (2019-02-12): Update libpng to version 1.6.36

Fix #3131
closed 2019-02-12 by nfeske