genodelabs/genode #3132: replace Post_signal_hook with a post-I/O-signal progress handler

ehmry, 2019-01-25:
The Post_signal_hook mechanism introduced with 6e5f3d432e8296c750bbd2e982f65ac543db1be0 is not robust when there is contention between different code-paths that wish to schedule a post-signal hook. A better mechanism may be to associate a single procedure and context with an entrypoint to be called after every batch of I/O signal dispatching. This relieves VFS plugins from explicitly scheduling such a hook, and avoids contention on what the hook body should be. VFS plugins should instead use the I/O response mechanism to schedule application activity, which would be presumed to be processed in the entrypoint I/O progress handler.

ehmry, 2019-01-28:
@nfeske as a clarification, should this handle be invoked after dispatching normal signals and I/O signals, or just I/O signals?

nfeske, 2019-01-28:
As far as I can see, only I/O signals are of interest. I cannot think of a situation where the execution of such a handler after regular signal handlers would be beneficial. So I'd propose to just call it after dispatching I/O signals.

ehmry, 2019-02-06:
I have a patch to implement this at ae9251be1595001abf8557b339a371740623305e. The patch removes the `Vfs::Io_response_handler` from the Libc and uses the post I/O signals hook instead, though I intend to use the `Vfs::Io_response_handler` mechanism again in the future.

nfeske, 2019-02-12:
I like your patch @ehmry. There is only one (cosmetic) change, I'd suggest. Could you rename `function` to `handle_io_progress`? This makes its purpose at both the call and implementation sites more obvious.

ehmry, 2019-03-01:
I have a new patch with an `handle_io_progress` interface at dbe814b384e0f864949c37f828d98c45be6f40a9, please add to staging.

nfeske, 2019-03-01:
I have the suspicion that the patch changes the API of the base system. @ehmry, to dispel my fear, could you check if the binaries produced with this patch in place remain compatible with the base system of 19.02?

If not, we should take the deliberate decision for or against breaking the ABI before publishing the official Sculpt CE images in mid of March.

ehmry, 2019-03-01:
Why is better to break the ABI just before release? Why not after?

ehmry, 2019-03-01:
Either it works or I don't know how to test if it works.

chelmuth, 2019-03-04:
I merged the first commit to staging. What's now missing IMO is the adaption of the file-system and terminal VFS plugins.

chelmuth, 2019-03-05:
The commit seems to break vfs_lxip, for example `make run/fetchurl_lxip`.
```
[2019-03-05 04:50:49] [init -> fetchurl] Using DHCP for interface configuration.
[2019-03-05 04:50:49] [init -> fetchurl] queue_delayed_work: queue_delayed_work called (from 0x16b54ef) return 0
[2019-03-05 04:50:49] no RM attachment (READ pf_addr=0x470 pf_ip=0x1190d21 from pager_object: pd='init -> fetchurl' thread='ep')
[2019-03-05 04:50:49] page fault, pd='init -> fetchurl' thread='ep' cpu=0 ip=0x1190d21 address=0x470 stack pointer=0xa04fef18 qualifiers=0x4 irUwp reason=1
```
@ehmry please have a look.

ehmry, 2019-03-05:
Vfs_lxip is triggering a `Libc::resume_all` during the construction of the Libc::Kernel singleton, patch at 
8cff5643f54c13439b0c5d997a3c0dc00027439d.

chelmuth, 2019-03-19:
The commit entered master. What are the next steps (commit was not marked as fix)?

ehmry, 2019-03-19:
I will close this and open another issue when the deprecated post-signal-hook can be removed.

closed 2019-03-19 by ehmry