genodelabs/genode #3138: dde_linux: iterate over PCI devices in ascending BDF order

cproc, 2019-01-31:
On a Lenovo ThinkCentre M57p, the system locks up when the UHCI controller BIOS handoff (disabling bit 4 in the LEGSUP register) for the controller with PCI BDF 00:1d:2 is attempted before the handoff for the controller with BDF 00:1a:0. The problem does not occur on Linux, where the handoff is done in ascending order of the BDFs, but on Genode the iterating order is currently reversed and the problem triggers.

nfeske, 2019-02-05:
How about moving the content of the excellent commit message as comment into the code (at the place where append is called instead of insert). This way, we won't accidentally violate this small but important semantics in the future.

chelmuth, 2019-02-05:
Done in bf2aceefa2d3752613135c2f7f28edd58bc3ceae

commit 6210bbd9 (2019-02-01): dde_linux: iterate over PCI devices in ascending BDF order

On a Lenovo ThinkCentre M57p, the system locks up when the UHCI controller
BIOS handoff (disabling bit 4 in the LEGSUP register) for the controller
with PCI BDF 00:1d:2 is attempted before the handoff for the controller
with BDF 00:1a:0.

Fixes #3138
commit 369d60bc (2019-02-12): dde_linux: iterate over PCI devices in ascending BDF order

On a Lenovo ThinkCentre M57p, the system locks up when the UHCI controller
BIOS handoff (disabling bit 4 in the LEGSUP register) for the controller
with PCI BDF 00:1d:2 is attempted before the handoff for the controller
with BDF 00:1a:0.

Fixes #3138
closed 2019-02-12 by nfeske