genodelabs/genode #3141: Removal of VirtualBox 4

nfeske, 2019-02-06:
All regular users of Genode's VirtualBox port use version 5 since a long time now. Version 4 is still sitting there because this is the only version supported on Muen. For me, however, it presents a maintenance burden. E.g., I'm in the middle of removing the use of deprecated Genode interfaces throughout the code base. Many changes I make for VirtualBox 5 must also be backported and tested for VirtualBox 4. I wonder, is this worth the effort?

@Kensan and @reet- What is your opinion? As far as I can see, VirtualBox 4 isn't too attractive for you as well but the adaptation of VirtualBox 5 is too costly to pursue. I guess you are eying on our plan to support VT with the base-hw kernel? Would you give us green light to drop VirtualBox 4 now even if that means that VirtualBox won't be usable on Muen until base-hw can be used as a nested virtualization layer on Muen?

reet-, 2019-02-07:
On 2/6/19 3:21 PM, Norman Feske wrote:
> @Kensan and @reet- What is your opinion? As far as I can see, VirtualBox 4 isn't too attractive for you as well but the adaptation of VirtualBox 5 is too costly to pursue. I guess you are eying on [...]

@nfeske: We agree to your assessment of the situation with vbox 4/5 and
Muen. Please got ahead with the removal of VirtualBox 4.

nfeske, 2019-02-11:
Thanks for the prompt feedback, @reet- .

@alex-ab Would you please go forward by removing the version 4 (removal of the code, port, run scripts, update of the autopilot list)?

alex-ab, 2019-02-11:
@nfeske: i will do, based on staging I assume ?

nfeske, 2019-02-11:
Yes.

alex-ab, 2019-02-11:
@nfeske: addressed by 61f6507bdbd0df91db098d74f90c47d801d083bc

nfeske, 2019-02-11:
Very good. Thanks @alex-ab for providing the commit so quickly.

alex-ab, 2019-02-11:
@nfeske: please remove the commit and use 29101a3207a8db4fce02953261688db16c87b84f - some files got lost

nfeske, 2019-02-12:
@alex-ab I merged your second commit to staging.

commit 026157cb (2019-02-12): Remove virtualbox 4 vmm

Fixes #3141
commit 48f10b5f (2019-02-12): Remove virtualbox 4 vmm

Fixes #3141
commit 6bb145bd (2019-02-19): Remove virtualbox 4 vmm

Fixes #3141
closed 2019-02-19 by nfeske