genodelabs/genode #3146: Soio5 packages and launcher for Sculpt

ehmry, 2019-02-07:
I'd like to create some packages and launchers for hosting user provided Solo5 unikernel images on Sculpt. In my first iteration I have a package that provides a dummy Solo5 images that prints out a diagnostic message and a seperate package that can load a user binary and serve a Block session from the file-system. The solo5 launcher would then depend on a loader launcher then depends on chroot. The launchers are documented to guide someone through replacing the unikernel binary and service routing.

ehmry, 2019-02-07:
The big TODO here seems routing through the nic_router NAT...

nfeske, 2019-02-07:
I like the idea. But I think we should not add more preinstalled launchers to sculpt. I rather like to reduce the number of launchers. My plan for the immediate release 19.02 is to enable the user to create a launcher for a given pkg interactively (via the '+' menu). With this mechanism in place, Sculpt wouldn't need most launcher examples anymore. How do you feel about merging the patches excluding the launchers?

nfeske, 2019-02-07:
Reviewing the launchers again, I must admit that the interactive creation tool most probably won't suffice to create those launchers because they contain more than just a few session routes (e.g., <config>). So it may be the best to cover them in a tutorial that describes the process of creating them.

ehmry, 2019-02-28:
This can be done with indexes now.

closed 2019-02-28 by ehmry