genodelabs/genode #3148: Terminal lacks unicode support

ehmry, 2019-02-08:


ehmry, 2019-02-08:
Fixed at 4c62beb5448fbd82ab0ada9137f409aaaed4dbb8. I can configure `input_filter` to send characters like ß and μ, but Bash will not echo these back without configuring the LC_BLAH_BLAH mush. Characters are rendered correctly when `cat` is run without arguments.

My motivation for these changes is making AZERTY keyboard work as expected #3149. I used these regularly in Berlin, but have yet to see one in Dresden.

nfeske, 2019-02-08:
Good to see that the `Codepoint` fits the terminal like a glove. The addition of `Codepoint::print` makes sense. I'm not 100% sure what to think of using UTF-32 for the cell array. To me this seems wasteful, when considering the addition of a scroll buffer in the future. I think in practice UTF-16 is appropriate for keeping internal representations of characters. AFAIK, Windows OS does it that way. I wonder, what does UFT-32 bring to the table for the terminal? Granted, I see the appeal of using the `Codepoint` type as is, but I also see that cell attributes and color inflate the size per cell in addition to the character. The use of UFT-16 would keep the cell to 32-bit.

Please keep the init config of noux-system as a separate file. It is large enough so that I'd like avoid its content being inlined into the config/managed/runtime. Inline runtime configs should be used only if they are small.

ehmry, 2019-02-08:
I think UTF-16 is reasonable (it even includes Braille), anything higher probably won't look right in a terminal anyway.

ehmry, 2019-02-08:
Fixup to use 16-bits internally and to revert the depot packages at 2f04adabd9d26ace870a03df3966a1349d8071f1.

ehmry, 2019-02-13:
I have a fixup to make the terminal_mux server build again at 57354c7f73a41a761e33aa98dd20b4bcc6768338, but its still missing unicode support.

ehmry, 2019-02-19:
I have a patch to add some unicode to the terminal_crosslink test at edfd74e332748c81d9c38b08871fab816acf9490.

nfeske, 2019-02-19:
The original commit went to master. I re-opened the issue for the follow-up adjustments.

ehmry, 2019-02-19:
The biggest outstanding issue that I am aware of is terminal_mux, which writes characters to ncurses.

nfeske, 2019-02-19:
Merged to staging.

nfeske, 2019-02-28:
Fixed in Genode 19.02.

commit 005a4c5e (2019-02-12): Refactor terminal for intrinsic Unicode support

Refactor the graphical terminal server to internally represent
characters as 16-bit codepoints and handle the duplex terminal stream as
UTF-8.

- Make the Codepoint class printable to the Output interface
- Decode data received at the Terminal session from UTF-8 to a 16-bit
  character
- Pass 16-bit characters through terminal decoder and char-cell arrays
- Send Unicode through terminal session in a burst of UTF-8 bytes

Fix #3148
commit 22327b43 (2019-02-19): Refactor terminal for intrinsic Unicode support

Refactor the graphical terminal server to internally represent
characters as 16-bit codepoints and handle the duplex terminal stream as
UTF-8.

- Make the Codepoint class printable to the Output interface
- Decode data received at the Terminal session from UTF-8 to a 16-bit
  character
- Pass 16-bit characters through terminal decoder and char-cell arrays
- Send Unicode through terminal session in a burst of UTF-8 bytes

Fix #3148
closed 2019-02-28 by nfeske