genodelabs/genode #3149: Sculpt lacks AZERTY support

ehmry, 2019-02-08:
Sculpt is missing support for the [AZERTY keyboard](https://en.wikipedia.org/wiki/AZERTY), both in the pre-defined character generators for `input_filter` and in the terminal unicode handling. Maybe in the font as well.

ehmry, 2019-02-12:
Patch at ff69d32a871e3c057978b72962eecb6af223fb52. I would refer to this as *azerty* and not *fr* because I have buiit the chargen using key labels on a keyboard that is not described as specific to any regional variant.

chelmuth, 2019-02-12:
Not quite according to https://en.wikipedia.org/wiki/AZERTY#Differences_between_the_Belgian_and_French_layouts_of_the_AZERTY_keyboard

ehmry, 2019-02-12:
Well its french then, I put the `+=}` key on the top row, french style.

ehmry, 2019-02-12:
I'm worried about getting in an argument with someone french about the details...

nfeske, 2019-02-18:
I have not quite followed the back and forth. ;-) Should the patch be cherry-picked to staging as is, or are there any adjustments planned?

ehmry, 2019-02-18:
I have no further improvements, but perhaps the name should be changed to *fr.chargen*.

nfeske, 2019-02-19:
@ehmry Is "perhaps" a yes or a no? ;-)

ehmry, 2019-02-19:
Yes, for consistency lets name it *fr*, patch for staging at 8a62870d77ee5922752b585bfc8856d310415f90.

nfeske, 2019-02-19:
Merged the fr-commit to staging.

commit d4cda0d8 (2019-02-19): Chargen description of French keyboard

Add an input_filter character description file for the French AZERTY
keyboard layout and include it with Sculpt.

Fix #3149
commit ffeb8e2a (2019-02-26): Chargen description of French keyboard

Add an input_filter character description file for the French AZERTY
keyboard layout and include it with Sculpt.

Fix #3149
closed 2019-02-26 by chelmuth