genodelabs/genode #3154: Trace::timestamp() causes exception on foc_pbxa9/QEMU

cproc, 2019-02-12:
@skalk and me debugged the previously mentioned (#3080) 'Trace::timestamp()' problem on foc_pbxa9 today, since lwip uses 'Trace::timestamp()' for its random number generator and all tests using lwip currently fail on this platform.

It turned out that the Fiasco.OC kernel assumes that performance monitors are not supported on this platform, because QEMU does not report a supported 'coprocessor debug model version' in the 'ID_DFR0' register for Cortex-A9. For this reason the kernel does not enable user mode access for the performance monitor registers and 'Trace::timestamp()' causes an exception when trying to read the cycle count register. QEMU actually implements the cycle count register and we could let the kernel always enable the user mode access as long as we don't support Fiasco.OC on an ARM platform without performance monitors.

cproc, 2019-02-12:
Commit [1] lets Fiasco.OC always enable the user mode access.

[1] https://github.com/cproc/foc/commit/cf42e69097ae5f1f166fd33390937d607229e181

cproc, 2019-02-14:
Commit c57aef7 updates the Genode port file to refer to my updated foc branch.

nfeske, 2019-02-14:
Thanks for the hint. I merged your commit to staging.

commit 0f7b5700 (2019-02-14): foc: always enable user mode access for performance monitors

QEMU implements the cycle count register read by
'Genode::Trace::timestamp()', but does not report a supported debug model
version for Cortex-A9.

Fixes #3154
closed 2019-02-19 by nfeske