genodelabs/genode #3155: base: timer driver utility only fits 32-bit

skalk, 2019-02-14:
The template function `timer_ticks_to_us` in `repos/base/include/drivers/timer/util.h` does not work properly for non-32bit types. You can test it, by trying the following code-snippet on a 32bit platform:
```
  enum { FREQ = 33333 }; 
  uint64_t tics = 0xffffffff;
  uint64_t ret = timer_ticks_to_us(tics, FREQ);
```
The return value will be zero although it should be ~128850307.

skalk, 2019-02-14:
The problem is the recurring wrong pattern of doing such casts:
```
  (RESULT_T)~0UL
```
Obviously, here a 32-bit type of value 0xffffffff gets casted to the result type (e.g. 64-bit). What is actually intended is having the maximum value of the result type. I've found the same misleading casts already in the hw-kernel at some points of the timer subsystem.

m-stein, 2019-02-14:
Hm, it is indeed a bit of a surprise to me that (unsigned long)~0UL should not produce 0xffff'ffff'ffff'ffff with 64 bit. When I run something like ``log("Test ", Hex(~0UL));`` (on x86_64 nova qemu) it yields ``Test 0xffffffffffffffff``.

skalk, 2019-02-14:
Yeah, on a 64-bit machine UL is 64-bit, on a 32-bit machine it is 32-bit.

skalk, 2019-02-14:
If your result type shall be 64-bit, you cannot use unsigned long as generic term, but e.g.: `~(RESULT_T)0`

m-stein, 2019-02-14:
Thanks for the explication and the commit :+1:

nfeske, 2019-02-14:
Thanks @skalk for tracking down this issue. I merged your commit to staging.

commit 36911a54 (2019-02-14): base: fix timer_ticks_to_us casting error

Fix #3155
closed 2019-02-19 by nfeske