genodelabs/genode #3157: hw: schedule on demand

skalk, 2019-02-14:
Until now, the model of the kernel was not tracking whether it is necessary to to make a new scheduling decision and thereby re-program the timer. Instead this was done whenever the kernel was executed (page-faults, syscalls, irqs, etc.).
Surely, this is a performance issue. In discussion with @nfeske, he thought it might be a cause for the heavy time drift on hw (see issue #3081). Actually, this was not the root-cause but introduced additional burden, which worsens the time drift.
Anyway, it is much better to: just inform the scheduler whenever a thread/vm got blocked or runnable again, or when a timer interrupt occurred. Then, invoke the scheduler and timer just once at the end of a kernel pass when there is the need to.

nfeske, 2019-02-14:
I merged your commit to staging, my small (inline) question notwithstanding.

chelmuth, 2019-02-15:
I reverted the commit over the weekend as there's no fix for the strange effects currently.

skalk, 2019-03-05:
@chelmuth @nfeske: I've identified and solved the remaining issue with riscv. The commit is ready for another run on staging.

chelmuth, 2019-03-05:
Merged the commit to staging.

commit 1155ccc2 (2019-02-14): hw: schedule on demand (Fix #3157)
commit fc7de8e2 (2019-02-15): Revert "hw: schedule on demand (Fix #3157)"

This reverts commit 1155ccc28fd5d8228cc494e467f65b97bc1410c3.
commit 83301e7a (2019-02-20): hw: schedule on demand (Fix #3157)
commit 24a543d2 (2019-02-22): Revert "hw: schedule on demand (Fix #3157)"

This reverts commit 83301e7aed59f26e947ddb9c0bd44d0514a83e57.
closed 2019-03-19 by chelmuth