genodelabs/genode #3158: Expose internal statistics of Block-backed VFS plugins

ehmry, 2019-02-14:
It would be nice to have an easy way to find out how much space is available in a file-system provided by Rump. It seems more reasonable to provide this informatin via synthetic files within the VFS rather than to open a report connection.

After some thinking and offline discussion I think the way to implement this securely and without side effects at the application is to create two directories within the plugin namespace, *content* and *control*, the former for the disk file-system and the latter for a *stats* file. This allows utilities that monitor disk usage to be partitioned from file-system content, and would not pollute the disk file-system with synthetic files.

The drawback is that an application would not be able to load rump into the VFS root directory and read and write without being aware of the semantics of the *content* directory.

nfeske, 2019-02-14:
How do you think about naming the `content` after the VFS node as I did for the TTF file system? See https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/raw/fonts_fs/fonts_fs.config for an example config. E.g., the /fonts/title/regular is a directory that contains the actual content.

A VFS node for rump would look as follows (the `name` attribute is new):

```
<vfs>
  <rump name="music" fs="ext2fs" ram="48M" writeable="yes"/>
</vfs>
<default-policy root="/music" />
```

Now, with a meaningful name "music" in place, we could name the meta directory after the name, prefixed with `.` - that would be `.music/`.

By following this convention, the libc can consistently ioctl operations for such directories. E.g., the libc would know that `.music/free` or `.music/avail` contains the information needed when `statfs` is called for `music/`.

@cnuke This discussion is very much related to ours about the ioctl handling for the terminal winsz, or block-device size. What do you think?

ehmry, 2019-02-14:
Yes that sounds better, it would be more flexible and allow mutiiple plugins be loaded into the same directory.

cnuke, 2019-02-14:
@nfeske Fine with me.

chelmuth, 2019-02-14:
Am I mistaken or does the `name` attribute not address @ehmry's original remark about the root directory of the VFS instance? In other words, `<rump>` creates always a sub-node for the content. Do we drop this feature?

nfeske, 2019-02-14:
> Do we drop this feature?

I would have no problem with dropping it. The only hypothetical downside is that it becomes more difficult to union-mount multiple file systems into one place.

@chelmuth If you feel uncomfortable with dropping the ability to mount a rump fs directly into a directory (or at /), we could also make the behavior depend on the presence of the `name` attribute. In the latter case, no meta directory would be provided. @ehmry Would that be feasible?

ehmry, 2019-02-14:
It would be feasible, but it would be simplier to implement only one behavior. The libc_vfs test in fact works more or less unchanged because the test assumes that it writes into */tmp*. 

Initial implementation at c3293fdac95ab7a4873ecea9b4535d51a2e43a80.

nfeske, 2021-02-26:
This issue has become stale. Sooner or later, we will come back to it but not in the foreseeable future.

closed 2021-02-26 by nfeske