genodelabs/genode #3160: ssh/tcp lags on hw/imx6

throwException, 2019-02-15:
I'm running the `ssh_terminal` runscript on the hw kernel on an imx6q and experiencing lags in the SSH/TCP connection. Some packets (keystrokes) seem to get processed arrive with a delay of ca. 1 second. The problem does not exist on hw/x86.

Wireshark shows that even the response to the first TCP SYN packet (the SYN+ACK) takes around 1 second. The SYN retransmission however is answered only a few milliseconds after the first SYN.

The packets seem to arrive in the with normal latency in the fec nic driver as well as lxip stack (dde_linux/src/lib/lxip/nic_handler.cc, _packet_avail) and once the answer packet leaves the lxip stack at (dde_linux/src/lib/lxip/nic_handler.cc, net_tx). The stack however seems to hang for ca. 1 second regularly.

Modified runscript: https://github.com/throwException/genode/blob/imx6_ssh_terminal/repos/gems/run/ssh_terminal.run

throwException, 2019-02-15:
Can someone confirm this problem also exists on hw/imx6q_sabrelite?

skalk, 2019-02-15:
@throwException well, I can confirm that we have serious timing issues on hw and especially CortexA9 which includes imx6 - see issue #3081. This of course can also effect a TCP/IP stack when timing is involved. Maybe, you can test my timing reqork as soon as it hits staging, which should be quite soon.

skalk, 2019-03-07:
@throwException the timing rework is now on staging. If you like, you might give it a try. Moreover, I've seen that the vfs_lxip stack stucks across different platforms, at least the test-libc_connect_lxip test. I'll investigate this issue currently and will inform you when I get more insights.

skalk, 2019-03-07:
@throwException my last remark was only related to the hw kernel, other kernels are not harmed

nfeske, 2021-01-25:
This issue has become stale since a long time. May we close it?

closed 2021-02-26 by nfeske