genodelabs/genode #3161: Write abritrary printable objects into Xml_generator

ehmry, 2019-02-15:
It would be convenient to write arbitrary objects as text into XML nodes, such as:

```
generator.node("number", [&] () { generator.append_sanitized(1U); });
generator.node("gain", [&] () { generator.append_sanitized(0.2); });

Genode::String<64> title("내 마음에 주단을 깔고");
generator.node("tile", [&] () { generator.append_sanitized(title); });
```

An interface like this may also be a bit safer.

ehmry, 2019-02-15:
Patch at 634efc54e7898f53cc04b25de070e1846060e386.

nfeske, 2019-02-17:
Very good idea and implementation!

The change of the test leaves me wondering a bit. Your patch reduces the set of covered characters from all possible byte values to just the few values listed in `pool`. I am not able to get the reasoning behind this change from the patch.

Generally, I prefer adding new test steps over changing the existing ones. By adding a new test step, we don't risk accidentally reducing the scope of testing.

ehmry, 2019-02-18:
I tried to keep the test of all byte values, but once a null byte is encountered the copy stops and the test result is short, which seems reasonable to me.

nfeske, 2019-02-18:
A null byte should not stop the append_sanitized but rather be sanitized. Think about the marshaling of arbitrary binary data into an XML node. This should just work.

ehmry, 2019-02-18:
I have a new patch at 1c7992c2273e59c9b8c1919da25b04025ebd0c99 that leaves the current behaviour and adds a new `content` method so that `content("count: ", (size_t)n)` is not ambiguous with `append_sanitized(s, strlen(s))`.

nfeske, 2019-02-18:
I like this much better. But how about calling the method `append_content`? This would express the semantics of the method better: When called multiple times, the previous content is not overwritten (as mistakenly indicated by a "setter" method like `content`) but the new content is appended to the existing content.

ehmry, 2019-02-18:
Ok, new patch at 9b7fc13f3409a79d4e9882b9a8261df586e8e359.

nfeske, 2019-02-18:
Perfect! Merged to staging.

commit a8f83286 (2019-02-18): Write abritrary printable objects into Xml_generator

Fix #3161
closed 2019-02-19 by nfeske