genodelabs/genode #3164: vbox5: remove use of deprecated APIs

nfeske, 2019-02-17:
There are still a few places in VBox5 that rely on deprecated APIs that I'm going to remove now. In particular the following compile errors occur when removing the `Signal_dispatcher` API:

```
/.../repos/ports/src/virtualbox5/devxhci.cc:328:10: error: ‘Signal_dispatcher’ in namespace ‘Genode’ does not name a template type
  Genode::Signal_dispatcher<Destruction_helper> dispatcher {
          ^~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/devxhci.cc: In function ‘int usb_signal_thread_wakeup(PPDMDEVINS, PPDMTHREAD)’:
/.../repos/ports/src/virtualbox5/devxhci.cc:364:56: error: ‘struct Destruction_helper’ has no member named ‘dispatcher’
  Genode::Signal_transmitter(pThis->destruction_helper->dispatcher).submit();
                                                        ^~~~~~~~~~
/.../repos/base/mk/generic.mk:56: recipe for target 'devxhci.o' failed
```
and
```
/.../repos/ports/src/virtualbox5/network.cpp:90:11: error: ‘Signal_dispatcher’ in namespace ‘Genode’ does not name a template type
   Genode::Signal_dispatcher<Nic_client> _link_state_dispatcher;
           ^~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:91:11: error: ‘Signal_dispatcher’ in namespace ‘Genode’ does not name a template type
   Genode::Signal_dispatcher<Nic_client> _rx_packet_avail_dispatcher;
           ^~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:92:11: error: ‘Signal_dispatcher’ in namespace ‘Genode’ does not name a template type
   Genode::Signal_dispatcher<Nic_client> _rx_ready_to_ack_dispatcher;
           ^~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:93:11: error: ‘Signal_dispatcher’ in namespace ‘Genode’ does not name a template type
   Genode::Signal_dispatcher<Nic_client> _destruct_dispatcher;
           ^~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp: In constructor ‘Nic_client::Nic_client(Genode::Env&, PDRVNIC, const char*)’:
/.../repos/ports/src/virtualbox5/network.cpp:170:4: error: class ‘Nic_client’ does not have any field named ‘_link_state_dispatcher’
    _link_state_dispatcher(_sig_rec, *this, &Nic_client::_handle_link_state),
    ^~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:171:4: error: class ‘Nic_client’ does not have any field named ‘_rx_packet_avail_dispatcher’
    _rx_packet_avail_dispatcher(_sig_rec, *this, &Nic_client::_handle_rx_packet_avail),
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:172:4: error: class ‘Nic_client’ does not have any field named ‘_rx_ready_to_ack_dispatcher’
    _rx_ready_to_ack_dispatcher(_sig_rec, *this, &Nic_client::_handle_rx_ready_to_ack),
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:173:4: error: class ‘Nic_client’ does not have any field named ‘_destruct_dispatcher’
    _destruct_dispatcher(_sig_rec, *this, &Nic_client::_handle_destruct),
    ^~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp: In member function ‘void Nic_client::enable_signals()’:
/.../repos/ports/src/virtualbox5/network.cpp:185:25: error: ‘_link_state_dispatcher’ was not declared in this scope
    _nic.link_state_sigh(_link_state_dispatcher);
                         ^~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:186:41: error: ‘_rx_packet_avail_dispatcher’ was not declared in this scope
    _nic.rx_channel()->sigh_packet_avail(_rx_packet_avail_dispatcher);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp:187:41: error: ‘_rx_ready_to_ack_dispatcher’ was not declared in this scope
    _nic.rx_channel()->sigh_ready_to_ack(_rx_ready_to_ack_dispatcher);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/ports/src/virtualbox5/network.cpp: In member function ‘Genode::Signal_context_capability Nic_client::dispatcher()’:
/.../repos/ports/src/virtualbox5/network.cpp:193:61: error: ‘_destruct_dispatcher’ was not declared in this scope
   Genode::Signal_context_capability dispatcher()   { return _destruct_dispatcher; }
                                                             ^~~~~~~~~~~~~~~~~~~~
/.../repos/base/mk/generic.mk:64: recipe for target 'network.o' failed
```

This issue is closely related to #3163. @alex-ab Could you please take care of fixing it?

nfeske, 2019-02-17:
BTW, the patch for removing the deprecated APIs can be found on my merge_to_staging branch (https://github.com/nfeske/genode/commits/merge_to_staging).

chelmuth, 2019-02-18:
I suggest @cnuke or @ssumpf look into this.

cnuke, 2019-02-18:
I'm already on it.

nfeske, 2019-02-19:
Fixed at master. Thanks @cnuke for the prompt response!

closed 2019-02-19 by nfeske