genodelabs/genode #3167: Configure buffer size of VFS File_system client at runtime

ehmry, 2019-02-18:
The JDK class loader was having problems because it was not checking for short reads caused by a small packet buffer at the VFS File_system client. This is a quick fix for this issue, a deeper investigation should be made in to the semantics of `read` and `fread`.

ehmry, 2019-02-18:
Patch at 7b48a7220d1d37597ca937c081cf82d017fc72e7.

nfeske, 2019-02-18:
That looks good - almost. ;-) If you wrap the default buffer size into a `Number_of_bytes` object, one could use the K/M/G suffixes in the attribute value. Would that be worth adding, @ehmry?

ehmry, 2019-02-18:
Yes, thats what I would prefer, new patch at 6ce7c2d1dbde65bbc8d95e54a1e344789b83ee1d.

nfeske, 2019-02-18:
Thanks for the new version. Merged to staging.

commit 2e05d269 (2019-02-18): VFS: Configure File_system client buffer at runtime

Make the size of the packet buffer at the VFS File_system client
configurable. This allows larger reads to be completed in a single
packet cycle.

Example: <vfs> <fs buffer_size="512K"/> </vfs>

Fix #3167
closed 2019-02-19 by nfeske