genodelabs/genode #3168: extend ability to add boards to Genode from outside (world)

skalk, 2019-02-18:
To add additional boards to Genode's ecosystem is not possible in general, although there are some exceptions that can already be used in genode-world, which are different Zynq boards.
Anyway, some useful information like the final linker address, or the base-library's specific package name are currently hardcoded within the run-tool plugin of the corresponding kernel. It would be nice to more easily add, e.g., a new ARM board to base-hw without the need to host them all in the central genode repository.

chelmuth, 2019-02-19:
For the problem at hand, which is support for boards in base-hw, the changes look appropriate. But beyond that, we need the notion of _board_ for all base platforms and depot archives too (see #2190 and a first attempt in 8620d0268449a6503a90420b86640b9e44acebbc). Could you imagine to broaden the scope of this patch series in this regard?

skalk, 2019-02-19:
Yes, actually I had this in mind by adding a board-specific file analoque to the kernel-specific file. For instance we could move procedure `board` to such files.

nfeske, 2019-02-19:
I'd not consider this change for the release 19.02. For the integration of the nitrogen board, please be as non-invasive as possible.

nfeske, 2019-02-20:
@skalk I merged the commits of your merge_to_staging branch to staging.

@ssumpf Could you test your scenario against this version?

nfeske, 2020-04-23:
@skalk Now that the code for the first couple of platforms has been moved to world, would now be a good time to close this issue?

skalk, 2020-04-23:
Yes, you're right.

closed 2020-04-23 by skalk

nfeske, 2021-01-19:
While documenting the pine64 porting process, I stumbled over some inconveniences for hosting the board support in an external repository. The trouble is mainly caused by the use of the REP_DIR variable in the *.mk files of base-hw. I have created the patch https://github.com/nfeske/genode/commit/c604e95b2401ecbd5cb37bc166dc95a1eb855248 to reduce the friction. @skalk and @m-stein would you be fine with this change? The patch is quite big but this is only because I tried to apply the patterns consistently to all boards.

skalk, 2021-01-19:
@nfeske I'm fine with fixing this problem once and for all, instead of doing it every time again, when somebody add a new board. Therefore, I'm very happy that you attend to it!