genodelabs/genode #3172: Sculpt: introduce index files for announcing packages

nfeske, 2019-02-19:
The current version of Sculpt comes equipped with some launcher configuration for installing additional software. However, for installing software beyond the scope of those launchers, a Sculpt user must know where to look and how to write launchers. A Sculpt user can easily miss cool features like https://genodians.org/ehmry/2019-01-31-flif_capture. In the future, software providers should become able to announce new packages to Sculpt users in a more convenient and formal way.

My proposed solution is the addition of the notion of index files to the depot. An index is named after the version of the matching Sculpt release. E.g., for Sculpt 19.02 this would be `depot/<user>/index/19.02`. This way, ABI incompatibilities between different Sculpt releases can be handled quite nicely. When a new version comes out, no index files exist. But once the packages of a software provider are updated, a new index (for the new Sculpt version) appears with the up-to-date versions.

An index file is an XML file like this:

```
<index>
  <index name="Games">
    <pkg name="Gambatte" path="ehmry/pkg/gambatte/2019-02-28" info="Gameboy emulator"/>
  </index>
  <pkg name="Nano 3D" path="nfeske/pkg/nano3d/2019-02-20" info="Software rendering demo"/>
</index>
```

The following points are worth noting:
* `<index>` nodes can be nested, which is intended as a way to structure the packages thematically.
* The content of `path` attributes may contain other authors than the one who maintains the index file. This allows anyone to create manual curated collections of software without providing the actual packages. E.g., after a new Sculpt release, anyone can step up and test packages of an older release and whitelist them in a custom index file when their work. This can be done independent from the original provider of the package.
* The description provided by the `info` attribute is meant to be short, adequate for text displayed in a tooltip.

In contrast to depot archives, which are immutable, index files are meant to be updated.

Sculpt will provide a mechanism to download such index files and construct the content of the `+` menu out of this information.

nfeske, 2019-03-01:
Fixed in Genode 19.02

closed 2019-03-01 by nfeske

tomga, 2019-03-11:
I'm not sure it is a good place to ask (especially it is an already closed issue), but could you explain how do you think index files are supposed to be used in context of different architectures? At first I would expect to see index files somewhere inside bin directory of depot to be able to have different index file for each architecture but index directory is directly inside depot and there seems to be no place to specify architecture. Neither in path or filename nor in the index file itself.

nfeske, 2019-03-11:
@tomga that's a good question - thanks for asking. At the current time, however, I don't have a good answer. Although the architecture question briefly crossed my mind, I admittedly disregarded it for the initial version because I failed to quickly come up with a satisfying idea how to handle it, and there were so many other pressing topics to solve.

I think that we should address this for the next version (around May).

Do you happen to have an idea how you would like to see the distinction implemented?

tomga, 2019-03-11:
I think my knowledge about depots and packages is not big enough to make me feel comfortable with proposing anything (I failed to start wm, decorator and layouter with current sculpt and I'm waiting for documentation :-) ). But I can write my observations - maybe they'll help.

I think that main point is that there has to be a possibility to have different "effective" indexes for different architectures. It's just a requirement I think.

I think the problem can be viewed from two perspectives:
1. client sculpt system asking for index file for depot
2. managing index information

For point 1 it seems that having specific index file for each supported by depot architecture is the most straightforward way. Those files could be stored in subdirectories similar to those in bin. It is the simplest solution from the client perspective which is I think important. Client knows own architecture and asks for index dedicated for him.

For point 2 some depot tools (publish?) can be used for helping with managing those index files for clients it at some point it is needed. If more information is shared between different architectures than local tool could process some common index "database" and produce effective index files dedicated for architecture. I don't expect that will be needed soon but it will be simple to add without breaking any compatibility.

I suspect that you, @nfeske, notice some potential problems with such simple approach as you probably already considered something like this.

nfeske, 2019-03-12:
@tomga thank you for sharing your ideas. I like your suggestion and will try to implement it in the next version. What was keeping me back was the second part of point 2 (the hassle of maintaining different index files for different architectures). But after all, this can be of course solved with tooling on the publishing side. Here, the work flow is still not completely thought through anyway.

With your suggestion, it comes down to choosing whether to store the index files as `depot/<user>/index/<arch>/<version>` or `depot/<user>/<version>/<arch>`. To be consistent with `bin/`, I would probably prefer the former.

Do you agree that this issue can wait for the next release?

tomga, 2019-03-12:
I understand that in the latter variant you omitted ``index`` by mistake? Anyway consistency is good so the former is probably better.

About the time of implementation I'm definitely not in the position to define deadlines :-). Anyway this can be important for me when I progress with my rpi3 work far enough to have sculpt running and I can't really estimate when it happens - at each level there are things that stop me unexpectedly :-) Probably next release will be ok for me and in case I need it earlier I'll ask or implement this change myself.

nfeske, 2019-03-12:
> I understand that in the latter variant you omitted index by mistake?

Sure. Sorry about that. ;-)

> Probably next release will be ok for me and in case I need it earlier I'll ask or implement this change myself.

I'm happy then. It is so cool to have the prospect to run Sculpt on the rpi3!