genodelabs/genode #3180: Consistently use the drivers_network-<board> pkg in run scripts

nfeske, 2019-02-22:
As an important step towards solving #2190s, we should remove the direct instantiation of `nic_drv` components from our run scripts but instead rely on `drivers_network` pkgs that provide an abstraction from the underlying hardware for scenarios that rely solely on networking.

By grepping the source tree for "nic_drv", one can see the list of potential run scripts to adapt. There are 34 matches in the Genode repository and about 10 in the world repository. The `drivers_network` pkg would fit for 28 of the run scripts of the Genode repository because those do not rely on any driver other than networking.

As a side effect of converting the run scripts, new `drivers_network` pkgs must be created in addition to the existing one for x86. In particular, we should introduce such pkgs for all boards that undergo our nightly tests.

I foresee that the result of this work will substantially reduce the hardware-specific content of our run scripts. In particular the conditional use of the usb_drv and platform_drv considerations would go away for the most part. After this step, it will be much easier to rename the actual driver components because this change will then stay local to the `drivers_*` pkgs but won't touch many run scripts.

Note that I'm not aiming at covering all run scripts with this approach. There will always be run scripts that are special (e.g., the virtualbox run scripts). But they are very few.

@skalk You may consider working on this topic for pursuing the ambition of unifying build directories across boards.

skalk, 2019-03-08:
@nfeske sorry for the late reaction, I missed this issue not by intention. Yes, you're right that this is an important step towards the unification goal. I take this on my radar.

skalk, 2019-05-03:
@nfeske @chelmuth to review and/or apply the commit chain accordingly, please look at my topic branch named "issue#3180". It consists of interleaved commits referencing either issue #2190 and this one, and the order is somehow crucial.

chelmuth, 2019-05-06:
Merged the series just now.

commit e72fb75c (2019-05-06): run: use driver_nic-* pkg where possible (fix #3180)
commit ebb0cbf2 (2019-05-15): fixup "run: use driver_nic-* pkg where possible (fix #3180)"
commit 7acd0337 (2019-05-16): run: use driver_nic-* pkg where possible (fix #3180)
commit acac2ed7 (2019-05-16): fixup "run: use driver_nic-* pkg where possible (fix #3180)"
closed 2019-05-29 by chelmuth