genodelabs/genode #3184: init: dies on uncaught Invalid_session exception

alex-ab, 2019-02-27:
Seen during usage with Sculpt:

```
[runtime] Warning: top-view: no valid RAM quota defined
[runtime] Error: Uncaught exception of type 'Genode::Pd_session::Invalid_session'
[runtime] Warning: abort called - thread ep
```

https://github.com/genodelabs/genode/blob/master/repos/os/src/init/child.cc#L229

seems to be the reason.

alex-ab, 2019-02-27:
Don't know whether a more sensible handling can be done, if such a exception happens at this point ...

nfeske, 2019-02-27:
Thanks for reporting.

I suspect that the problem lies elsewhere. I wonder why the PD cap of the child can be invalid in the first place. It shouldn't. So this needs further investigation.

Can you reproduce the problem in a simple scenario, e.g., by using one of our existing run scripts with a slight modification?

alex-ab, 2019-02-27:
Unfortunately it can only be reproduced on Sculpt up to now. Whenever I shutdown a VM and afterwards try to interact with Leitzentrale, init reliable dies. According to the messages of the kernel the VM got definitely killed, so that means also the Pd_session cap should be invalid?! In the list of children however the vbox child is still listed, according to instrumentation, and during downgrade()/upgrade(). https://github.com/genodelabs/genode/blob/staging/repos/os/src/init/child.cc#L293

```
--- a/repos/os/src/init/child.cc
+++ b/repos/os/src/init/child.cc
@@ -239,6 +239,8 @@ void Init::Child::_apply_resource_upgrade(QUOTA &assigned, QUOTA const configure
 
 void Init::Child::apply_upgrade()
 {
+       warning(name(), ": downgrade");
+
        if (_resources.effective_ram_quota().value == 0)
                warning(name(), ": no valid RAM quota defined");
 
@@ -250,6 +252,8 @@ void Init::Child::apply_upgrade()
 
        _apply_resource_upgrade(_resources.assigned_cap_quota,
                                _configured_cap_quota(), _cap_limit_accessor);
+
+       warning(name(), ": downgrade done");
 }
```

```
[runtime] inspect: downgrade
[runtime] inspect: downgrade done
[runtime] vbox5-nova-sculpt: downgrade
[runtime] Error: Uncaught exception of type 'Genode::Pd_session::Invalid_session'
[runtime] Warning: abort called - thread ep
```

Does this info help you ?

chelmuth, 2019-02-27:
In https://github.com/genodelabs/genode/blob/staging/repos/os/src/init/main.cc#L446 and https://github.com/genodelabs/genode/blob/staging/repos/os/src/init/main.cc#L453 operations are skipped for *abandoned* children. I suspect this also has to be done for `apply_upgrade()`. @nfeske noted that it may be a good idea to move this *abandoned-guarding* into the actual functions.

alex-ab, 2019-02-27:
The vbox client is not abandoned - so this suggestion does not help.

```
diff --git a/repos/os/src/init/child.cc b/repos/os/src/init/child.cc
index ce3eeb0df0..3253732e59 100644
--- a/repos/os/src/init/child.cc
+++ b/repos/os/src/init/child.cc
@@ -239,6 +239,8 @@ void Init::Child::_apply_resource_upgrade(QUOTA &assigned, QUOTA const configure
 
 void Init::Child::apply_upgrade()
 {
+       warning(name(), ": upgrade ", abandoned());
+
        if (_resources.effective_ram_quota().value == 0)
                warning(name(), ": no valid RAM quota defined");
 
@@ -250,6 +252,8 @@ void Init::Child::apply_upgrade()
 
        _apply_resource_upgrade(_resources.assigned_cap_quota,
                                _configured_cap_quota(), _cap_limit_accessor);
+
+       warning(name(), ": upgrade done");
 }
 
 
@@ -292,6 +296,8 @@ void Init::Child::_apply_resource_downgrade(QUOTA &assigned, QUOTA const configu
 
 void Init::Child::apply_downgrade()
 {
+       warning(name(), ": downgrade ", abandoned());
+
        Ram_quota const configured_ram_quota = _configured_ram_quota();
        Cap_quota const configured_cap_quota = _configured_cap_quota();
 
@@ -322,6 +328,8 @@ void Init::Child::apply_downgrade()
                               "cap_quota=", demanded_cap_quota};
                _child.yield(args);
        }
+
+       warning(name(), ": downgrade done");
 }
```
```
[runtime] vbox5-nova-sculpt: downgrade 0
[runtime] vbox5-nova-sculpt: downgrade done
...
[runtime] inspect: downgrade 0
[runtime] inspect: downgrade done
[runtime] vbox5-nova-sculpt: upgrade 0
[runtime] Error: Uncaught exception of type 'Genode::Pd_session::Invalid_session'
[runtime] Warning: abort called - thread ep
```

alex-ab, 2019-02-27:
So, just to repeat me - this vbox client stops itself (send a exit notification to init) - it is not killed via the leitzentrale.

alex-ab, 2019-02-27:
c96a452af7fd101298f0f1de43a052a6e8e009bd fixes the issue.

nfeske, 2019-02-27:
Super, Alex! Thanks for thorough investigation. I merged your fix to staging.

commit 6d70e555 (2019-02-27): init: avoid upgrade of resources on exited child

Fixes #3184
closed 2019-02-28 by chelmuth