genodelabs/genode #3192: trace: support more thread related information

alex-ab, 2019-02-28:
Possible directions:
- execution time by a thread-context (foc, seL4, future: nova)
- execution time by a scheduling-context (nova)
- calculated quantum, period (today: hw, future: nova)
- priority (various kernels)
- kernel memory usage (future: nova)

alex-ab, 2019-02-28:
On my [branch 3192](https://github.com/alex-ab/genode/tree/issue_3192) the proposed changes regarding thread-context, scheduling-context, quantum and priority can be found.

nfeske, 2019-03-25:
Reviewing https://github.com/alex-ab/genode/commit/47e3675b30cab25ea05da15789933797901c8732, shouldn't the kernel-memory usage info go to a data structure different from `Execution_time`?

I'm actually not quite convinced that the kernel memory should be associated with a trace subject. Kernel memory is accounted to a PD whereas a trace subject corresponds to a thread.

To export kernel-memory stats to the user land, could we possibly use a ROM module instead?

alex-ab, 2019-03-29:
> To export kernel-memory stats to the user land, could we possibly use a ROM module instead?

Actually, that was the first version (outdated nowadays), see https://github.com/alex-ab/genode/commits/nova_core_pd_stat_dump. But I have some doubt whether this is something you want to have upstream.

A general support for kernel specific values is in my opinion is more desirable, e.g. in case of seL4 where you can also specify this value per pd.

But I'm fine to host this feature just for me. The other commits are more interestingly actually, we may skip this kernel memory for the moment.

alex-ab, 2019-03-29:
I adjusted the priority/quantum commit to provide values for sel4/hw/foc as far as available.

nfeske, 2019-04-02:
@alex-ab I merged the four commits.

The PD-stats ROM looks quite useful, too. But you are right that the patch is quite invasive.

Maybe it could be simplified by removing the signalling but letting the client poll for new versions by calling `update` (the polling would be at the client side).

The `Object_pool::apply_all` approach with fishing out the `Pd_session_component` objects with dynamic cases is quite brutal. It would be nicer to keep the PD sessions in a dedicated registry.

An issue may arise if multiple clients request the same ROM. Each client would need its own copy because each client calls `Rom_session::update` at its own pace. One statically allocated `Rom_module` won't cut it.

For the time being, let's keep this handy tool on a topic branch.

alex-ab, 2019-04-02:
@nfeske: thanks

nfeske, 2019-04-03:
The depot autopilot reports some failures that may be related to this issue. In particular, test-trace and test-trace_logger on OKL4 are affected.

alex-ab, 2019-04-03:
Please consider 153c593f3d9e81960ec651c70a2e4b3640786225 as quick workaround for the limited UTCB size of OKL4. As discussed off-line, the general direction would be to switch to a shared-memory/ROM approach in the future for requesting the Subject_info data.

nfeske, 2019-04-03:
Thanks for the fixup and the background info. I'd be glad about switching the mechanism over to the use of the dataspace as done for requesting the subject IDs.

alex-ab, 2019-04-10:
Most of the features entered master, closing it for now.

closed 2019-04-10 by alex-ab