genodelabs/genode #3193: Sculpt: allow for removal of index files

nfeske, 2019-02-28:
Currently, a Sculpt user can easily download a package index from a software provider but once downloaded, it is not easily possible to update it because the downloading mechanism finds the index file already present. To update an index, it first needs to be removed (from both the depot/ and the public/ directories). Right now, this can be done manually using the inspect window via
```
cd <sculpt-partition>
rm depot/<provider>/index/19.02 public/<provider>/index/19.02
```
After this manual step, a new version of the provider's index can be downloaded in the depot selection menu.

Admittedly, this is rather awkward. The process of updating an index would become much more convenient if Sculpt would allow the de-selection of items of the depot selection menu. When de-selected, the corresponding index files could be removed from the Sculpt partition. So for refreshing an index, the user would just need to toggle the checkbox twice.

nfeske, 2019-03-05:
As a stepping stone, I'm considering a new component called `fs_tool` that instantiates a VFS and accepts a list of operations in its configuration. E.g.,
```
<config>
  <vfs>
     <dir name="depot"> ... </dir>
     <dir name="public"> ... </dir>
  </vfs>
  <unlink path="/depot/nfeske/index/19.02"/>
  <unlink path="/public/nfeske/index/19.02.xz"/>
  <unlink path="/public/nfeske/index/19.02.xz.sig"/>
```

The same tool could also help to implement a saving feature for the current sculpt runtime configuration. So the user may perform the copying of /config/managed/deploy to /rw/config/19.02/deploy via a simple press of a button:

```
<copy from="/config/managed/deploy" to="/hw/config/19.02/deploy"/>
```

@ehmry What do you think of this?

ehmry, 2019-03-05:
I like it. This could also be used when preparing VM ISOs to avoid using Noux+coretuils.

NobodyIII, 2019-03-10:
I like the `fs_tool` idea as well. I thought of it in the past, but never got around to implementing it.

nfeske, 2019-03-10:
@NobodyIII thanks for expressing your interest in the tool. I have just taken this sub topic to a dedicated issue (#3222).

nfeske, 2019-03-12:
With the current staging branch, this works now.

commit a26efc15 (2019-03-12): sculpt: allow for removal of index files

By clicking on a yellow checkbox in the depot selection dialog, the
corresponding index files are removed. This way, index files can
be update by removing and downloading them again.

This patch also filters out sculpt-managed components from the graph to
avoid erratic graph-position changes while the '+' menu is open.

Fixes #3193
closed 2019-03-19 by chelmuth