genodelabs/genode #3196: Phase out the use of the `Framebuffer` and `Input` sessions

nfeske, 2019-02-28:
I plan to remove the today's notion of framebuffer and input sessions as part of my work on revisiting Genode's low-level GUI interfaces (planned in the road map for 19.05).

Both sessions will be replaced by new sessions that work in the reverse direction. I.e., nitpicker would no longer connect to the framebuffer and input drivers but the drivers would connect to nitpicker. This is a prerequisite for the ability to restart drivers, the sharing of one screen among multiple drivers, the creation of screen-capturing utilities, multi-head support, and proper headless support for Sculpt.

Before I can go ahead into this direction, however, we need to successively reduce the impact of such a profound change by migrating existing components away from the to-be-removed interfaces. Generally, all framebuffer clients should become nitpicker clients. As nicely illustrated by @ehmry's recent work on https://github.com/genodelabs/genode-world/pull/138, this change potentially simplifies many scenarios.

@alex-ab VirtualBox and Seoul are important components that spring in mind immediately. Would you take on the work of changing those?

Another change that would play into my hands would be the removal of manually instantiated framebuffer and input drivers from run scripts by converting run scripts to the use of `drivers_interactive` depot pkgs. This already happened for most run scripts in gems.

alex-ab, 2019-02-28:
> @alex-ab VirtualBox and Seoul are important components that spring in mind immediately. Would you take on the work of changing those?

Sure, i will take care.

alex-ab, 2019-03-05:
@nfeske: in commit 3117fbd1a18ffd36153c1d5c34051d40a7ccc689 and a4f50f153137f9212d01e667db7955072d500a29 Seoul and Vbox5 are now using Nitpicker, instead of Framebuffer + Input Session separately

chelmuth, 2019-03-05:
I'm uncertain to merge the commits before Sculpt CE. Opinions?

nfeske, 2019-03-05:
As the commits don't change interfaces, I don't see a big risk about Sculpt CE.

However, we should consider use cases where VBox is currently deliberately be combined with nit_fb for policy reasons at least draft a rough proposal of how those use cases could be covered now. E.g., policy could be imposed by a bump-in-the-wire nitpicker-to-nitpicker component in place of today's nit_fb. I'm not saying that we should implement this. But we should be able to pull a reasonably solid answer to such questions out of our sleeve.

alex-ab, 2019-03-05:
I would not merge them before Sculpt CE.

nfeske, 2019-03-05:
@alex-ab Thank you for the commits. Good to see that quite a bit of code could be removed in the vbox commit.

nfeske, 2019-03-25:
Now with Sculpt CE out of the door, I merged @alex-ab's commits for VBox and Seoul to staging.

nfeske, 2019-03-27:
The vbox tests `vbox5_genode_usb_hid` and `vbox5_win7_64_share` suffer from this change. @alex-ab may you please have a look?

```
[2019-03-27 05:49:13] [init] [34mWarning: virtualbox: no route to service "Nitpicker" (label="virtualbox -> ")[0m[0m

[2019-03-27 05:49:13] [init -> virtualbox] [31mError: Nitpicker-session creation failed (ram_quota=36864, cap_quota=9)[0m[0m

[2019-03-27 05:49:13] [init -> virtualbox] [31mError: An unexpected exception occurred[0m[0m

[2019-03-27 05:49:13] [init -> virtualbox] [31mError: MachineWrap : Failed to assign the machine to the session (NS_ERROR_FAILURE)[0m[0m

[2019-03-27 05:49:13] [init -> virtualbox] main     [0m

[2019-03-27 05:49:13] [init -> virtualbox] main     !!Assertion Failed!![0m

[2019-03-27 05:49:13] [init -> virtualbox] main     Expression: SUCCEEDED(getObjectState().getState() == ObjectState::InUninit || [0m[0m
```

alex-ab, 2019-03-27:
I'm already on it.

alex-ab, 2019-03-27:
@nfeske: please add a585b719cc8c653a88ca29c797d764d1da24f34e to staging

nfeske, 2019-03-27:
Done. Thank you.

nfeske, 2021-02-26:
All the important aspects of this issue are solved. There are a few remaining users of the framebuffer/input sessions (liquid_fb, gui_fb) but those are minor edge cases.

closed 2021-02-26 by nfeske