genodelabs/genode #3198: Update Arora default bookmarks

nfeske, 2019-02-28:
The default bookmarks of the Arora web browser are embarrassingly outdated. I propose to change them to the following list (in this particular order):

* https://genodians.org
* https://genode.org
* https://github.com/genodelabs/genode
* https://lists.genode.org/pipermail/users/
* https://reddit.com/r/genode
* https://www.genode-labs.com

Are there any further suggestions or opinions? E.g., what about @cnuke's hardware-compatibility list, or Genode's Wikipedia article, or microkernel.info, or the respective kernel's project pages (seL4, NOVA, Muen, Fiasco.OC), or Genode-covering news sites (phoronix.com, pro-linux.de)?

nfeske, 2019-02-28:
BTW, the bookmarks are defined at https://github.com/genodelabs/genode/blob/master/repos/ports/src/app/arora/patches/arora_bookmarks.patch

cproc, 2019-03-12:
Sounds good.

commit 2f926e19 (2019-03-19): Update Arora default bookmarks

Fixes #3198
commit 9f50b889 (2019-04-01): Update Arora default bookmarks

Fixes #3198
closed 2019-04-02 by nfeske