genodelabs/genode #3199: select board specific drivers when building depot archives

trimpim, 2019-03-01:
While building package archives for a imx6 (arm_v7a) the binaries for the drivers weren't built. This is due to the fact, that in the spec.conf file that is created to build the binaries only contains arm_v7a as spec but not imx6.

I have a small patch, that I use currently, but I'm not sure, if this is the way to go forward, especially regarding the ideas in #2190 and #3180.
It requires, that I specify an additional environment variable (BOARD) that then is added to the specs.conf file

nfeske, 2019-03-01:
The use of specs for differentiating boards will be eliminated in the future.

It's good that you found a stop-gap solution for your case but this won't be suitable for upstream. A particular problem of this approach is that binary archives built for the same depot path but with different `BOARD` arguments would show up at the depot under one and the same name but with different content. This ambiguity calls for trouble.

A better solution is to strip out the `REQUIRES` declaration from the `target.mk` file in the `content.mk` of your driver package. Have a look at https://github.com/genodelabs/genode/blob/master/repos/os/recipes/src/pbxa9_drivers/content.mk for an example.

trimpim, 2019-03-04:
Thanks for the information, on how this can be handled.

closed 2019-03-04 by trimpim