genodelabs/genode #3200: Window layouter: permit floating windows to resize themselves

nfeske, 2019-03-01:
Right now, the window layouter enforces that window size never exceeds the bounds as defined by the user, which is the desirable behavior when using tiled windows. However, for foating windows, the application knows best, what window size is appropriate. Right now, the application can specify the initial window size but it cannot change it after the fact. Whenever the application resizes the window's nitpicker view, the window can only become smaller but not larger.

To fix this issue, the window layouter should become less strict about enforcing the user-define window size when a window is floating.

@alex-ab I have noticed that this effect is very much in the way of your new version of top_view when routing it to the window manager.

nfeske, 2019-03-13:
Fixed by https://github.com/genodelabs/genode/commit/0609285c19ad26701bf3fb107765427f708f6c1e on staging.

nfeske, 2019-03-13:
Unfortunately, my fix interacts badly with issue #3227. With the fix in place, a user-resized window snaps back to the original window position. :-/

alex-ab, 2019-03-14:
> @alex-ab I have noticed that this effect is very much in the way of your new version of top_view when routing it to the window manager.

Thanks for the heads up. I will take with the next update of top_view.

nfeske, 2019-03-14:
Commit https://github.com/genodelabs/genode/commit/7430541a39c829977f3cc79a755c794fb149f821 (on staging) fills the gap I mentioned in my previous comment.

commit 0609285c (2019-03-13): window layouter: allow floating apps to resize

This patch gives applications the ability to control the size of their
window whenever the window is floating, not tiled or maximized. See the
comment in the code for the rationale.

Fixes #3200
closed 2019-03-19 by chelmuth