genodelabs/genode #3202: Invalid session label change

jklmnn, 2019-03-04:
While implementing a block server I noticed that the session label seems to be changed by constructing a `Constructible<Attached_ram_dataspace>`. I created a reproducer at https://github.com/jklmnn/genode/commit/e0cba3a516add1e71eb3be192dba949268e10539 that can be run with `make run/block_server`.

When creating a session I took the session label:
```C++
const char *label = Genode::session_label_from_args(args.string()).last_element().string();
Genode::log("Label addr=", (void*)label, " len=", Genode::strlen(label), " (", __FILE__, ":", __LINE__, ")");
if (ds_size >= ram_quota.value) {
    Genode::warning("communication buffer size exceeds session quota");
    throw Genode::Insufficient_ram_quota();
}

Genode::log("Label addr=", (void*)label, " len=", Genode::strlen(label), " (", __FILE__, ":", __LINE__, ")");
_ds.construct(_env.ram(), _env.rm(), ds_size); //Constructible<Attached_ram_dataspace>
Genode::log("Label addr=", (void*)label, " len=", Genode::strlen(label), " (", __FILE__, ":", __LINE__, ")");
_session.construct(_env.rm(), _ds->cap(), _env.ep(), _request_handler, _heap); //Constructible<Block_session_component>
Genode::log("Label addr=", (void*)label, " len=", Genode::strlen(label), " (", __FILE__, ":", __LINE__, ")");
```
(https://github.com/jklmnn/genode/blob/e0cba3a516add1e71eb3be192dba949268e10539/repos/os/src/test/block_server/genode_block_server.h#L73-84)

The expected output is that all logs provide the same output. But when running the test the length of the label is changed:
```
[init -> test-block_server] Block server
[init -> test-block_server] Label addr=0xa01fea60 len=0 (/.../repos/os/src/test/block_server/genode_block_server.h:74)
[init -> test-block_server] Label addr=0xa01fea60 len=0 (/.../repos/os/src/test/block_server/genode_block_server.h:80)
[init -> test-block_server] Label addr=0xa01fea60 len=4 (/.../repos/os/src/test/block_server/genode_block_server.h:82)
[init -> test-block_server] Label addr=0xa01fea60 len=4 (/.../repos/os/src/test/block_server/genode_block_server.h:84)
[init -> test-block-client] block device with block size 512 sector count 1024 (testing 1024 sectors)
[init -> test-block_server] Label addr=0xa01fea60 len=0 (/.../repos/os/src/test/block_server/genode_block_server.h:74)
[init -> test-block_server] Label addr=0xa01fea60 len=0 (/.../repos/os/src/test/block_server/genode_block_server.h:80)
[init -> test-block_server] Label addr=0xa01fea60 len=1 (/.../repos/os/src/test/block_server/genode_block_server.h:82)
[init -> test-block_server] Label addr=0xa01fea60 len=4 (/.../repos/os/src/test/block_server/genode_block_server.h:84)
[init -> test-block-client] reading block 0 - 1023, 246 per request
```
This happens on NOVA and Linux (I didn't check the other kernels). Interestingly when the `test-block-client` opens its second session the change happens twice.

chelmuth, 2019-03-04:
In https://github.com/jklmnn/genode/blob/e0cba3a516add1e71eb3be192dba949268e10539/repos/os/src/test/block_server/genode_block_server.h#L73 you create a `Session_label` temporary and store a cstring pointer into `label`. After this source-code line, the temporary is invalid (ie, lost) and `label` is a dangling pointer pointing into stack space that may be used by the successive function calls.

jklmnn, 2019-03-04:
Does `last_element` also create another temporary?

chelmuth, 2019-03-04:
That is what I expect in this case: If I do not declare a storage location for the result of a function, it is only temporary (and thus no longer valid after the statement). I suggest you try the following which should not fail.
```c++
Genode::String label = Genode::session_label_from_args(args.string()).last_element();
Genode::log("Label addr=", (void*)label.string(), " len=", Genode::strlen(label.string()), " (", __FILE__, ":", __LINE__, ")");
```

jklmnn, 2019-03-04:
Well isn't in your case the result of `Genode::session_label_from_args(args.string())` also temporary? I'm aware that I cannot reuse function results if I don't store them but the semantics of the `Session_label` functions is not clear to me. If I understand your example correctly `last_element()` creates a new `Session_label` object?

chelmuth, 2019-03-04:
Okay, let me try to unravel it.

1. `Session_label::last_element()` returns a `Session_label` instance (which is a temporary until stored).
2. `Session_label` is a `String` and inherits the `String::String()` constructor as well as the `String::string()` function.
3. `String` contains a buffer and *imports* the string data on construction into the object's buffer.
4. `String::string()` returns a `char const *` into the objects buffer, which is unsafe for temporaries but valid for variables with a storage location until the variable is out-of-scope.

Also in your example code you never refer to `args` or `label` except from the debug output, which hints that the issue must be related to `label`. Please correct me if I oversee something.

chelmuth, 2019-03-04:
> Well isn't in your case the result of `Genode::session_label_from_args(args.string())` also temporary?

Yes this is correct: The return value is a temporary which is then stored into my `label` variable. Note, the compiler may optimize out the copy-construction step without `-fno-elide-constructors`. The difference of my approach is that I preserve the object while your code just stores a pointer into the temporary. In some cases, your code may actually work depending on the stack-allocation scheme of the compiler but still relies on an invalid assumption.

jklmnn, 2019-03-04:
Thanks for the thorough explanation. I'll close this ticket since this isn't a bug.

closed 2019-03-04 by jklmnn