genodelabs/genode #3204: Always set return value in clock_gettime(REALTIME|SECOND)

ehmry, 2019-03-04:
Since it has become a reoccurring problem that applications are not (ever?) checking the return value of calls like `gettimeofday`, perhaps a relative time should be placed in the return value regardless of whether or not the call will return an error because of an an unconfigured RTC.

ehmry, 2019-03-04:
Patch at 343fd6279b4876aa12c477033c9718b03f802c22.

chelmuth, 2019-03-07:
In the nightly CI we can now see that *extract* fails with a missing timer session. This is due to *libarchive* depending on Libc's `time()` in `archive_write_disk_new()`. The former implementation just returned `errno=EINVAL` in the test as it misses any RTC configuration, which could be fixed with the ` <inline name="rtc">2000-01-01 00:00</inline>` scheme. The current *time.cc* always depends on `Libc::current_time()` - either if RTC is configured to calculate the correct time stamp or if RTC is not configured to apply the above-mentioned safety net.

This brings the situation that we cannot prevent users of *libarchive* from using a Timer session. The question is now: Should we walk this path? For me we should relax the requirement for time-less components in this regard if we want to achieve increased compatibility/robustness of ported POSIX applications. If a use case requires a constrained notion of time for specific components, this could be applied on Timer session level.

nfeske, 2019-03-07:
@chelmuth I admittedly was not able to follow your last paragraph completely.

From a users point of view, I intuitively want to limit the reach of the libc as far as possible unless the user takes an informed decision (i.e., grant access to a timer session). There must be a plausible reason to take such a decision. For example, as a user, it is not plausible that a timer is needed to extract an archive. The output of the program should not depend in any way on time or timing. So a timer session should not be a requirement in this case.

That is at least my line of thinking. A different position would be to state that every libc-using program must have access to a timer session, stating the notion of time as a strict requirement for such POSIXy programs. This would of course accommodate the case of ported applications a little bit better. But on the other hand, it creates implausible requirements for simple applications like the extract tool.

I would really like to keep our credo of being as strict as possible by default, also with regard to handing out timer sessions.

chelmuth, 2019-03-07:
Then we had to change the libarchive implementation as it depends on the `time()` function of libc but according to your rationale does not need it.

nfeske, 2019-03-07:
This discussion takes quite a few twists now. ;-)

@ehmry How about returning a simple increasing counter as a fake time value in the fallback case? This would presumably solve the original problem as stated in the issue description while not changing any policy about timer sessions.

I am bit curious about the severity of the issue. In the cases where you ran into the problem, isn't there a warning printed about the attempt to access the rtc? If yes, couldn't one solve the problem by just configuring the rtc? Doesn't the existing warning already pinpoints the problem well enough? I'm sure that I'm missing something.

ehmry, 2019-03-07:
I don't see value in preventing a libc application from accessing the timer. As I understand it, the application is still free to keep time by polling the TSC in a loop. The problem is that I have yet to see any code that handles an error response from a time function.

nfeske, 2019-03-07:
@ehmry The value of restricting access to timing is certainly debatable. But I wonder, is this issue the right place for such a discussion? In my opinion, the answer should not be dictated by (in)conveniences experienced while porting applications.

You left my questions unanswered.

What actual problem did you hit and how severe was it?

Would a valid rtc configuration have side-stepped your problem in the first place?

Would the incremented counter as dummy value have avoided your problem?

ehmry, 2019-03-07:
My last issue was that Nim was using `gettime` to call an HTTP client progress callback once per second. By returning a time of zero or -1, broken-by-default, the callback was never called, because time was never advancing. This would have been avoided by an inline RTC file in the VFS, but I see this is counter-intiuitive, I didn't need a RTC time, just monotonic time.

I know that would fix it not because this is documented, but because I've seen this break many other things in the past.

nfeske, 2019-03-07:
In cases like this, one can usually use `<inline name="rtc">2000-01-01 00:00</inline>` as rtc configuration, which results in monotonic time.

Don't you agree that this explicit policy is preferable over a built-in fallback (that is just a heuristic after all)? Isn't this simple enough?

nfeske, 2019-03-07:
In the Nim example, was there a warning about the lacking rtc configuration? I'd hope so.

ehmry, 2019-03-08:
An RTC configuration is not an explicit fix for a broken timer, it is a stochastic fix.

nfeske, 2019-03-08:
Please bear with me. I'm not trying to be stubborn. I simply have not managed to grasp the problem. When I'm unable to see the problem, I cannot anticipate your solution. All I can see is the uncertainty that comes with changing things that seem to work fine now.

Please help me to put myself into your perspective so that I can empathize.

ehmry, 2019-03-08:
The problem is that `clock_gettime` is the backend for other time functions like `gettimeofday`, which may return an error, see http://pubs.opengroup.org/onlinepubs/9699919799/.

This bug is related to #3018, #2329, #3057, and problems with genodians. If I didn't know why `gettimeofday` could return -1 I would just assume that the genode libc is broken and not worth using.

nfeske, 2019-03-08:
@ehmry please let us continue this discussion in person. I'm afraid we keep running in circles otherwise.

commit 9b455ea9 (2019-03-05): Set return value of clock_gettime regardless of error

Applications are not checking for errors when using the time functions
of the libc. Return a monotonic time regardless if an error will be
returned due to an unavailable RTC or an invalid clock request.

Fix #3204
closed 2019-03-13 by ehmry