genodelabs/genode #3205: gcc configure script "fixes" headers from host system

NobodyIII, 2019-03-05:
When compiling noux-pkg/gcc_x86 on my computer, I ran into a mysterious error:

`tar: ./lib/gcc/x86_64-pc-elf/6.3.0/include-fixed/python3.7: File removed before we read it`

I found it particularly mysterious that python 3.7 would have anything to do with compiling GCC. I discovered that this error was a result of the GCC build process trying to fix any potentially problematic system headers on the host system, despite compiling for a different platform. I discovered that this problem occurs even when cross-compiling for ARM (noux-pkg/gcc_arm).

Even on host systems that compile GCC successfully, this bug leads to non-Genode headers ending up in Genode's GCC includes.

ssumpf, 2019-03-08:
@NobodyIII: Thanks for investigating. Can you please clarify what you mean by non-Genode headers? Are this headers, for example Python headers, Genode GCC does not need, or are headers overwritten or wrong headers included? In the meantime will check if the `tar` error occurs on my system.

NobodyIII, 2019-03-08:
By non-Genode headers, I mean headers from the host system, including
headers for libraries that haven't even been ported to Genode, e.g.
Irrlicht.

ssumpf, 2019-03-11:
@NobodyIII: I have found the following README in `./lib/gcc/x86_64-pc-elf/6.3.0/include-fixed` which may help to explain the problem:

`This README file is copied into the directory for GCC-only header files
when fixincludes is run by the makefile for GCC.`

`Many of the files in this directory were automatically edited from the
standard system header files by the fixincludes process.  They are
system-specific, and will not work on any other kind of system.  They
are also not part of GCC.  The reason we have to do this is because
GCC requires ANSI C headers and many vendors supply ANSI-incompatible
headers.`

`Because this is an automated process, sometimes headers get "fixed"
that do not, strictly speaking, need a fix.  As long as nothing is broken
by the process, it is just an unfortunate collateral inconvenience.
We would like to rectify it, if it is not "too inconvenient".`

Please note, that none of these headers actually make it into the noux `gcc_x86.tar` or `gcc_arm.tar` and are apparently required to compile the tool chain. So GCC might simply copy to many headers from the host system. Does this answer your questions? If so, I would like to close this issue.

NobodyIII, 2019-03-11:
The tar error came during the final packaging step from a fixincludes
symlink that pointed a nonexistent target, which happened because the
includes-fixed directory is part of the final install, contrary to your
claim. However, I discovered that the install-no-fixincludes target removes
these files, at the expense of not stripping the binaries. Unfortunately,
there is no built-in combination of install-no-fixincludes and
install-strip. I have a patch that fixes this by making install-strip use
install-no-fixincludes instead of install as its base. I'll submit it soon.

ssumpf, 2019-03-12:
@NobodyIII : Okay, my guess is that your host has a broken link in the system headers (see README above). It would be nice if you could specify  **exactly** what is installed to **exactly** where, as well as what you mean by **installed** (the gdb_arm.tar archive?), because I must be able to reproduce your issue.

nfeske, 2021-02-26:
Since this issue remained unattended for more than a year, I'm closing it.

closed 2021-02-26 by nfeske