genodelabs/genode #3209: sculpt: immediat hover update on menu-dialog changes

nfeske, 2019-03-05:
Sculpt's `+` menu changes under the mouse pointer whenever a sub menu is entered or left. Right now, the item of the new content on the existing pointer position is not hovered automatically. In order to click it, the user first needs to move the pointer a little bit, with makes the usage of the menu a bit notchy.

alex-ab, 2019-03-05:
Last week I came up with commit b62358e6e080a66bc82fa00b85685428aebcf02c to fix a issue which sounds like this for the top view package.

nfeske, 2019-03-06:
Thanks for the hint. Your commit looks close to what I intended to do, except for different naming of things. Would you mind if I take your commit as a basis and add my changes as a fixup?

nfeske, 2019-03-06:
Nevermind. I just noticed that it probably easier to apply my changes to the original state (as your `_check_hover_state` method contains more code than needed for the handle-config case).

nfeske, 2019-03-08:
The issue turned out to be much more subtle than expected. The three commits above improve the handling of corner cases in the menu view, the window manager, and the sculpt manager. Now the `+` menu feels much more responsive.

closed 2019-03-19 by nfeske

chelmuth, 2021-09-02:
I discovered a situation were a button is not clickable without moving the mouse in Sculpt:

1. Start a component from `+`, open its bubble dialog, and hover restart.
2. Switch to desktop and back by pressing `F12` twice.
3. Try to click the button without moving the mouse -> nothing happens.
4. Just nudge the mouse and click successfully.

I thought this observation might be related to this issue but could open a dedicated issue if desired.