genodelabs/genode #3212: test/bomb: Fix missing Timer-session error

trimpim, 2019-03-06:


skalk, 2019-03-06:
@trimpim thank you for the contribution, I've tested it and the ugly warnings vanished :-)

chelmuth, 2019-03-06:
Sorry but this commit completely changes the semantics of the test, which identifies the master bomb by a successful Timer-session creation. Please review locations in code which contain`if (timer.constructed())`.

trimpim, 2019-03-11:
@chelmuth I was not aware of this mechanism. To keep my automated test happy, I'd like to get rid of this message.

Would a attribute, that is passed from the parent to its children, be an acceptable solution?

chelmuth, 2019-03-11:
@trimpim after your initial patch I was also pondering on alternatives to the timer session based detection. The config ROM is certainly the best fitting resource to *configure* bomb instances as `master="false"` (with a  default of `true`). I'd happily review a patch for this.

trimpim, 2019-03-18:
@chelmuth what do you think about d7727be98f42022de95fee982208164fb6c10825

chelmuth, 2019-03-18:
The patch is quite what I had in mind. If you could please address my inline note and try to rephrase the commit message to something like *bomb: prevent Timer-session creation errors*. IMO the commit does not *fix* the test but beautifies the log ;-)

trimpim, 2019-03-19:
@chelmuth the updated commit is  at 245ea6645f8d0d9567a89ed980e327da88c63519

chelmuth, 2019-03-19:
Merged to staging.

commit b266b9f0 (2019-03-19): bomb: prevent Timer-session creation errors

Fix #3212
commit 8456ffa2 (2019-04-01): bomb: prevent Timer-session creation errors

Fix #3212
closed 2019-04-02 by nfeske