genodelabs/genode #3214: Investigate unsrepsonsiveness in long running lighttpd scenario

cnuke, 2019-03-06:
The Genodians.org appliance, i.e. the lighttpd component, suffers from being unreachable from the outside after some time.

cnuke, 2019-03-11:
So far the investigation is somewhat inconclusive as I tried to reproduce the problem locally but could not.

To recap the problem at hand: every now and than, the time interval differs, the genodians website cannot be reached from the outside anymore. Looking at the (scarce) LOG messages does not reveal anything eye-catching. During the first test-phase of the appliance we restarted the VM manually whenever it got “stuck”. By now a script running on the KVM host checks the availability of the website to detect any anomalies (i.e. `curl --head --max-time 10 …`). In case the server does not respond in time the VM gets restarted.

The architecture of the appliance is as follows: the KVM instance the genodians VM runs in is managed by libvirt and connected to a virtual switch. The traffic from the public IP is routed to the VM's private network by means of iptables. The host system is Debian and Qemu/KVM 2.8.1 is used.

I tried to reproduce to get the appliance to the point where it does not respond anymore locally, e.g. running the appliance in Qemu (w/ and w/o KVM). Since we still do not have any meaningful information (after all we do not log accesses, neither in the appliance nor on the KVM host) my first obvious idea was to put some pressure on the appliance by using a synthetic load-generator (in this case I used ab(1) as well as siege(1) with varying parameters). Depending on how the appliance was run, either Qemu (2.11 on Ubuntu 18.04) got stuck (*main-loop: WARNING: I/O thread spun for 1000 iterations*) or it crashed (*[…] kernel: traps: qemu-system-x86[7452] general protection ip:55f4add5c249 sp:7f191e53dc70 error:0 in qemu-system-x86_64[55f4ad79d000+8d2000]A*) repeatedly. Those issues I do not see on the Debian host where the VM is still running AFAICT.

At that point I switched to running the appliance on bare-metal directly. For better or worse there it worked fine, even when hit by a hugh load up to the point were lighttpd could not accept new connections anymore and accessing the website interactively triggered a timeout most of the time in the client (lynx). But as soon as I stopped the load generator, all went back to normal and I could access the website.

My assumption initially was that the appliance gets stuck, which is way I started to instrument the nic_router as well as the libc and thereby the vfs_lxip plugin. At least on bare-metal that does not seem to be the case.

So, to have a better starting point, I would like to enable logging on the Debian host (I have to figure out how to do that, though). I already did add a `date_log` component which prefixes all log messages in the genodian subinit with the current date to the appliance (after all doing it there is easier than to prefix Qemu's serial messages on the KVM host).

nfeske, 2019-03-12:
Thanks for the very interesting write-up!

Once the scenario "gets stuck", is the import subsystem also affected (it fetches content over the network after all) or is the problem local to the web server?

cnuke, 2019-03-12:
At least when looking through the LOG messages some weeks back the import subsystem seemed not be affected (which prompted me to look for the problem in the appliance rather than the surrounding environment) and to be honest I have not paid much attentation to that lately. That being said, I already took measures that will make the state of the appliances as a whole more obvious (hopefully) the next time it gets stuck.

cnuke, 2019-03-18:
Observing the latest restarts confirmed that the appliance is still running but the lighttpd component is the one that cannot be reached anymore at some point.

nfeske, 2020-04-17:
Fixed thanks to the thorough investigation by @chelmuth and the commit 6dae147785cfd0fd7293eedb5c3a5754c9520bba.

closed 2020-04-17 by nfeske