genodelabs/genode #3215: i.MX 6 Solox: Fetch stale TLB entry

ssumpf, 2019-03-06:
During tests I encountered illegal memory access from instructions like `blx lr`, which do not access memory at all. After some investigation I found out that the initial value of one faulting IP looked like this:
`Error: ip: 0x10e5ef0 opcode: 0xe320f000` (`nop` added by hand in library)
when the fault occurs the opcode at IP is this:
`Kernel: 1: ip 0x10e5ef0 opcode: 0xe5d13000 ptbr: 0x8030404a` (`ldrb r3, [r1]`).
Note: No updates to the page table at this range have been made. This let Stefan and me insert a TLB flush right after the `Kernel: 1` message and re-read the opcode at IP:
`Kernel: 2: ip 0x10e5ef0 opcode: 0xe320f000 ptbr: 0x8030404a` which ends up being correct (`nop`).

Therefore, we see a TLB hit from another address space, which should not happen because the kernel uses ASIDs. We tried to fix the issue with barriers, branch predictor flushing, and even TLB flushes during context switches without success. We also studied the erratas for the SoloX, which does have quite a few regarding RAM. In the next step we will try to use the diagnostic registers in order to reconfigure the SoC and use the values Linux programs into the board.

nfeske, 2019-03-06:
That looks scary.

ssumpf, 2019-03-07:
Note: 5e87aa1 is just for a demo scenario and should not enter staging at this point, since it decreases performance. Otherwise, it works just fine for the demo.

ssumpf, 2020-04-03:
Should be fixed with dad042d.

closed 2020-04-03 by ssumpf