genodelabs/genode #3217: vfs_lxip stucks reliable on all hw platforms

skalk, 2019-03-08:
At least the nightly tests regarding netperf with lxip and `test-libc_connect_lxip` reliably stuck across all different hardware on the hw kernel.
I've investigated that in the `test-libc_connect_lxip` the client successfully sends a packet through the nic_router to the server, but the server application never receives it, because it does not wait for signals. When instrumenting the kernel, I could trace that the server once wait for a signal and receives one, but never again calls `await_signal` syscall. Therefore, it obviously cannot receive a signal not packet.
Next I try to investigate the state of the server component, and why it never again calls the block_io_signal function. I expect a deadlock in the libc/vfs_lxip model. Due to a quite different runtime behaviour on hw (no global signal receiving thread), it would explain the occurrence only on this kernel. Other suggestions are welcome.

It is quite probable that issue #3160 is related to this one.

nfeske, 2019-03-08:
Thanks for sharing your intermediate analysis.

skalk, 2019-03-08:
Ok, it is no deadlock, but a lifelock. The server component's ep never blocks again from a certain point on. When sampling the ip of the server component ep thread, I get the following sample:
```
            0x4b970            0x4b970  ld.lib.so                       Genode::Avl_node_base::remove(Genode::Avl_node_base::Policy&)(T) +    0x0 = 0x4b970
                                                                                                      in file /home/sk/src/genode/repos/base/src/lib/base/avl_tree.cc:121
            0xd27a0            0xd27a0  ld.lib.so                       __gxx_personality_v0(T) +  0x186 = 0xd261a
            0xcb1cb            0xcb1cb  libc.lib.so                     getopt_internal(t) +  0x13b = 0xcb090
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/getopt_long.c:406
            0xcb1e1            0xcb1e1  libc.lib.so                     getopt_internal(t) +  0x151 = 0xcb090
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/getopt_long.c:424
            0x49950            0x49950  ld.lib.so                       Genode::Allocator_avl_base::_alloc_block_metadata()(T) +    0x0 = 0x49950
                                                                                                      in file /home/sk/src/genode/repos/base/src/lib/base/allocator_avl.cc:99
            0x830cd            0x830cd  ld.lib.so                       Genode::Heap::free(void*, unsigned long)(T) +   0x8d = 0x83040
                                                                                                      in file /home/sk/src/genode/repos/base/src/lib/base/heap.cc:243
            0xcb1a1            0xcb1a1  libc.lib.so                     getopt_internal(t) +  0x111 = 0xcb090
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/getopt_long.c:430
            0xa5c80            0xa5c80  ld.lib.so                       Genode::Slab::Block::_slab_entry(int)(T) +   0x20 = 0xa5c60
                                                                                                      in file /home/sk/src/genode/repos/base/include/util/misc_math.h:54
            0xd226e            0xd226e  ld.lib.so                       read_encoded_value(_Unwind_Context*, unsigned char, unsigned char const*, unsigned long*)(t) +   0x41 = 0xd222d
            0xd282a            0xd282a  libc.lib.so                     strtoul(T) +    0xa = 0xd2820
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/strtoul.c:49
            0xcab60            0xcab60  libc.lib.so                     parse_long_options(t) +    0x0 = 0xcab60
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/getopt_long.c:201
            0x49ac0            0x49ac0  libc.lib.so                     pthread_getspecific_int(t) +    0x0 = 0x49ac0
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/gen/_pthread_stubs.c:224
            0xd1d4f            0xd1d4f  libc.lib.so                     strtoimax(T) +   0x2f = 0xd1d20
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/stdlib/strtoimax.c:49
            0xa5c80            0xa5c80  ld.lib.so                       Genode::Slab::Block::_slab_entry(int)(T) +   0x20 = 0xa5c60
                                                                                                      in file /home/sk/src/genode/repos/base/include/util/misc_math.h:54
            0x4af48            0x4af48  libc.lib.so                     pthread_attr_setstackaddr_int(t) +   0x58 = 0x4aef0
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/gen/_pthread_stubs.c:260
            0xccc70            0xccc70  ld.lib.so                       _Unwind_GetRegionStart(t) +    0x0 = 0xccc70
                                                                                                      in file /home/krishna/tmp/genode.git/contrib/gcc-e062a6fcfe6dcbfcba8bbb58f833669766487b56/src/noux-pkg/gcc/libgcc/unwind-dw2.c:373
            0xd362c            0xd362c  ld.lib.so                       std::type_info::__is_pointer_p() const(t) +    0x0 = 0xd362c
            0x49c8d            0x49c8d  libc.lib.so                     pthread_mutex_destroy_int(t) +   0x3d = 0x49c50
[... cut ...] bc/gen/_pthread_stubs.c:260
            0x4af0c            0x4af0c  libc.lib.so                     pthread_attr_setstackaddr_int(t) +   0x1c = 0x4aef0
                                                                                                      in file /home/sk/src/genode/contrib/libc-be4908ff7037258395bb0dfbc5d4c5ffef22f859/src/lib/libc/lib/libc/gen/_pthread_stubs.c:260
            0xcae68            0xcae68  ld.lib.so                       execute_cfa_program(t) +  0x178 = 0xcacf0
```

skalk, 2019-03-08:
Ok, further analysis showed that in the special hw signal handling an exception gets thrown that is not used on other kernels at `repos/base/src/lib/base/signal_common.cc:195`. That exception never gets caught in the error-case of the server component. Normally it should get caught at `repos/base/src/lib/base/signal_common.cc:153`. The throwing seems to end up in an endless loop. Maybe another exception gets thrown during allocation of the actual exception?

skalk, 2019-03-08:
Yes `malloc` in the cxx library gets called again and again.

skalk, 2019-03-08:
My previous observation was half-cocked. It is indeed the case that the faulty component again and again throws an exception, but not because some weird cxx library path. Instead the Libc Entrypoint code calls over and over again the Signal_receiver::pending_signal function, but in non-blocking state. Therefore Signal_receiver::wait_for_signal never gets called, and there is no execution context in base-hw that can receive a signal from the kernel in that case.

skalk, 2019-03-11:
Ok, the final outcome is the twofold:
* On hw kernel, if there is a libc component actively polling for io signals without blocking somewhere in the meantime, io signals do not arrive, because there is no second thread putting signals in the corresponding registry like on other kernel. A fix is to ask the kernel for the signals without blocking.
* the VFS lxip plugin does not return the correct return value when accept is called on a socket. It returns READ_OK when it actually should indicate to block. This leads to an active loop as long there is no new connection available.

skalk, 2019-03-12:
@chelmuth there is a fixup commit for the signal changes in hw that solves the send/received signal differences on my merge_to_staging branch commit 48e311d1deb2118facd510663b6475a310d07890

commit 730f90f2 (2019-03-11): hw: introduce non-blocking signal checking

* Introduces pending_signal syscall to check for new signals for the
  calling thread without blocking
* Implements pending_signal in the base-library specific for hw to use the
  new syscall

Fix #3217
closed 2019-03-19 by chelmuth