genodelabs/genode #3218: sculpt CE: leitzentrale gui popup non functional after a while

alex-ab, 2019-03-08:
```
[leitzentrale -> gui -> popup] resource_request: ram_quota=377056
[leitzentrale -> gui] child "popup" requests resources: ram_quota=377056
```

alex-ab, 2019-03-08:
Probably related, in the log beforehand one can see:
```
[leitzentrale -> gui] Warning: popup assigned RAM (12M) exceeds available RAM (6596441)
[leitzentrale -> gui -> popup] tar archive 'menu_view_styles.tar' local at 0x158000, size is 102400
```

nfeske, 2019-03-08:
Does this occur depending on the time running (which would hint at a memory leak) or triggered by the popup becoming larger than before? I hope that it's the latter case so we can just tune the quotas.

ssumpf, 2019-03-08:
I had the same issue and could trigger it by copying `deploy` to `/config` and starting/configuring the `vbox5-nova-sculpt`. When choosing File System (wm) the popup would cause the resource request. Increasing the quota in `Sculpt::Gui::_gen_menu_view_start_content` did work for me and the behaviour has not occurred this week.

alex-ab, 2019-03-08:
Thanks for the hints, @ssumpf. After giving the popup the requested memory (12M), I could not reproduce it.

nfeske, 2019-03-08:
Thanks for the fix. Merged to staging.

closed 2019-03-19 by alex-ab