genodelabs/genode #3220: Audio Driver in VirtualBox VM Not Working

jjkarcher, 2019-03-08:
In a VirtualBox virtual machine (including Sculpt VC pre-packaged VM), when attempting to use the "cnuke/pkg/bsd_audio_drv" package (using either the instructions in "https://genodians.org/alex-ab/2019-02-13-sculpt-vc-audio" or the Sculpt CE "+" menu), the following error is logged:

[runtime -> bsd_audio_drv] "Error: could not initialize sound card"

There is no audio output when continuing the instructions to load "audio_player".

System Info:

VirtualBox Host: Debian Linux (stable)
VirtualBox Version(s): 5.22 and 6.04

VM Audio Card: Intel HD Audio

This should be easily reproducible, but if you need further details or want me to try anything, just let me know.

cnuke, 2019-03-09:
Thanks for the report, using the driver in a VM is “unsupported” for the time being, however.

When I ported the driver over from OpenBSD 5.7 initially I noticed that it did not work in VirtualBox. I tried to get audio to work in VirtualBox with an vanilla OpenBSD 5.7 guest, which I did not manage either. At that time I decided against debugging the issue as the driver worked fine on real hardware and using it in a VM was not a use-case.

With Sculpt that has changed as people are attempted to try it out in a VM rather than booting it on real hardware. So that issue is something we should considers when we “officially” support audio on Sculpt.

jjkarcher, 2019-03-27:
May I plead the case for officially supporting VirtualBox?

The main reason is that you can support any reasonably recent Intel/AMD machine by supporting just one hardware target.  In my case, I am running an AMD FX-series desktop and a 2009 MacBook Pro, and Genode runs beautifully in VirtualBox on both.

Since VBox has the "Intel HD Audio" option, one hopes that it's a relatively minor problem (although you never know until digging in, obviously).

jjkarcher, 2019-08-01:
I installed the latest (6.5) version of OpenBSD in a VM, to see what would happen with audio.  In short, the results were:
1. With the VM set for "Intel HD Audio", there was no sound.
2. With the VM set for "ICH AC97", the sound worked.

I guess that doesn't bode well for getting the current driver to work with the "Intel HD Audio" setting.

Just for curiosity, what might be involved in adding the AC'97 portion of the OpenBSD driver to the current driver?

cnuke, 2019-08-01:
Thanks for checking with a current version!

Adding the driver should be pretty straightforward as the AC97 stuff is already ported:

* Add the driver file *./dev/pci/auich.c* to *dde_bsd/audio.list* and prepare the port
* Add the file to `SRC_C` in *dde_bsd/lib/mk/dde_bsd_audio.inc*
* Hook the driver up to the autoconfiguration emulation code in *dde_bsd/src/lib/audio/bsd_emul.c*

Then try to compile the `audio_drv` component and if necessary extended the OpenBSD kernel emulation so that this step succeeds.

jjkarcher, 2019-08-23:
Sorry for clogging up the issue tracker with novice questions, but I am stuck in the build step.  Hopefully I am missing something simple.

I added "./dev/pci/auich.c" to "dde_bsd/audio.list".  Preparing the port was successful, and the "auich.c" file was downloaded into "genode/contrib/dde_bsd-641aa94a26f460745a56c3223f3aa3cfbea12dca/src/lib/audio/dev/pci/" along with the other files.

Then I modified "dde_bsd/lib/mk/dde_bsd_audio.inc" as you indicated, and took my best educated guess on "dde_bsd/src/lib/audio/bsd_emul.c".

Then I downloaded your "bsd_audio_drv" package, and tried to build it.  This succeeded, but didn't include any of my changes to "dde_bsd".

As an experiment, I made the same changes to "dde_bsd_audio.inc" and "bsd_emul.c" inside the "bsd_audio_drv" directory.  (There doesn't seem to be an "audio.list" in there.)  Trying to build this fails with "dev/pci/auich.o: No such file or directory".

Am I missing a step between preparing the "dde_bsd" port and building "bsd_audio_drv"?

cnuke, 2019-09-19:
You managed to follow my somewhat sparse hints, which is nice nonetheless.

I pushed commit 43d8296 that adds the ICH driver source and adds the dummies to compile the driver succesfully. I have not tried to run it but probably it will not work properly as at least the `auich_calibration()` function might not be happy with the current implementation of `microuptime()`. I guess that's pretty much in line with your changes.

And the commit c3f2786 adds the source file to the src archive recipe. The orginal src archive does not contain the file, mostly likely the cause for the error when you tried to build the package archive. That's indeed the missing step.

For the time being, I suggest to use the `audio_out.run` run-script to test and debug the driver, which does not uses packages yet, and run the resulting .iso in VirtualBox (or even on HW).

(Sorry for the late anwser, I was on vacation.)

jjkarcher, 2019-09-24:
Thanks for doing this!  (I guessed correctly on the boilerplate code, but I wouldn't have been able to figure out the timer part.)

When I run it, there is no sound, and just a blank screen.  I haven't run this scenario before - is there supposed to be any sign of life besides audio?

Is there a way to redirect the Log to a serial port (which VBox can store in a file)?

cnuke, 2019-09-26:
I appreciate that you spend your time looking into this topic.

I am afraid not, the run script will simply playback the *sample.raw* file and all output is printed to the LOG session but it is already connected to the serial console.

IIRC you can enable a serial adapter in the VM settings in the VBox GUI. Picking `COM1` and redirecting it to a file should do the trick (I recommend using a Linux host or similar instead of Sculpt for debugging the driver in VBox to have a proper working base-line regarding audio in VBox).

jjkarcher, 2019-10-02:
I am running Vbox on a Debian host, and I have COM1 redirected to a text file.  I've tried it several times, always with the same result: I get the "Genode" boot screen, then it goes blank, but I don't get any sound, and no COM1 log file is created.

For a comparison, I booted the same image on my new ThinkPad (that's another whole story), and it did make crackling noises.  That implies to me that I built the image correctly, but that my "sample.raw" file is not valid.  (I used "sox" as described in the README file.)

Are you aware of a known good "sample.raw" file that I can download?

Also, can you infer anything from the fact that there is no COM1 output in VBox at all?

cnuke, 2019-10-12:
I created a new *[sample.raw](https://usr.sysret.de/jws/genode/misc/sample.raw)* file from an MP3¹ (I had to install *libsox-fmt-mp3* on Debian) by following the instructions given in the README:

```
sox -c 2 -r 44100 Chad_Crouch_-_Shipping_Lanes.mp3  sample.f32
```

and confirmed that it played back properly using

```
play -c 2 -r 44100 sample.f32
```

I shortend the file to 4 MiB, which gives us about 12 seconds of audio

```
dd if=sample.f32 of=sample.f32 bs=4M count=1
```

The resulting *sample.raw* I checked with the *audio_out.run* script on *base-linux* (sounds fine) and on Qemu (sounds distorted but that is to be expected). I also copied the *audio_out.iso* to an USB stick and succefully executed it on my DS57U (using NOVA).

> For a comparison, I booted the same image on my new ThinkPad (that's another whole story), and it did make crackling noises. That implies to me that I built the image correctly, but that my "sample.raw" file is not valid. (I used "sox" as described in the README file.)

Could you please tell me which Thinkpad did you use? Although the HDA driver is known to work on all models I've tested so far, it is also possible that it does not properly work on your machine and your sample file is actually fine.

> I am running Vbox on a Debian host, and I have COM1 redirected to a text file. I've tried it several times, always with the same result: I get the "Genode" boot screen, then it goes blank, but I don't get any sound, and no COM1 log file is created.

> Also, can you infer anything from the fact that there is no COM1 output in VBox at all?

I'm afraid I don't kown why you don't get any output. I also tried to run the *audio_out.iso* in VirtualBox and I indeed get the serial output in the specified raw file.

Please check your *.vbox* file, mine contains

```Xml
<UART>
  <Port slot="0" enabled="true" IOBase="0x3f8" IRQ="4" path="/home/jws/audio_out.log" hostMode="RawFile"/>
</UART>
```

That being said, a look into the serial file reveals that the scenario does not work

```
Genode 19.08-115-g95baf5d7d9 <local changes>
210 MiB RAM and 63253 caps assigned to init
[init -> test-audio_out] --- Audio_out test ---
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 15
[init -> acpi_drv] XSDT OEM 'VBOX  ', table id 'VBOXXSDT', revision 1, creator 'ASL ' (97)
[init -> acpi_drv] SMBIOS table (entry point: 0x16ff60 structures: 0xe1000)
[init -> platform_drv] Error: ACPI report parsing error.
[init -> platform_drv] Error: Uncaught exception of type 'int'
[init -> platform_drv] Warning: abort called - thread: ep
[init -> audio_drv] --- probe drivers ---
```

but I've not investigated what is the reason for that.

¹) the first one that popped up on the Free Music Archive for *Instrumental*

jjkarcher, 2019-10-13:
Thanks for this info!

Your "sample.raw" file worked on my ThinkPad, so that eliminates one variable.  (I must have done something wrong with my conversion.)

I also must have had a permissions problem with my Log file directory, because I changed the path and now it shows similar (but not identical) output to yours above:

> Genode 19.08 <local changes>
> 2002 MiB RAM and 63253 caps assigned to init
> [init -> test-audio_out] --- Audio_out test ---
> [init -> acpi_drv] Found MADT
> [init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
> [init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 15
> [init -> acpi_drv] Found MCFG
> [init -> acpi_drv] MCFG BASE 0xdc000000 seg 0x0 bus 0x0-0x3f
> [init -> acpi_drv] XSDT OEM 'VBOX  ', table id 'VBOXXSDT', revision 1, creator 'ASL ' (97)
> [init -> platform_drv] ECAM/MMCONF range 00:00.0-3f:1f.7 - addr [00000000dc000000,00000000e0000000)
> [init -> audio_drv] --- probe drivers ---
> [init -> acpi_drv] SMBIOS table (entry point: 0x16ff60 structures: 0xe1000)
> [init -> platform_drv] Root bridge: 00:00.0
> [init -> platform_drv] 00:05.0 adjust IRQ as reported by ACPI: 11 -> 21
> [init -> platform_drv] 00:05.0 uses IRQ, vector 0x15, non-maskable
> no RM attachment (READ pf_addr=0x20 pf_ip=0x10299bc from pager_object: pd='init -> audio_drv' thread='ep') 
> page fault, pd='init -> audio_drv' thread='ep' cpu=0 ip=0x10299bc address=0x20 stack pointer=0xa04fef18 qualifiers=0x4 irUwp reason=1

I don't know if that's enough to track down the page fault - let me know if I can give you more details.

P.S. The ThinkPad is a L380 Yoga (see https://www.notebookcheck.net/Lenovo-ThinkPad-L380-Yoga-i5-8250U-FHD-Convertible-Review.295936.0.html).  As far as I can tell, it works with Genode except for a USB problem.  Now we know audio works, too.  :^)

cnuke, 2019-10-16:
*(FWIW, my .iso did not work because I forgot to set the system's chipset to `ICH9`. After doing so I saw the some page-fault you did.)*

Following the fault, I managed to get the driver to work in VirtualBox after the fixing shortcomings in the OpenBSD emulation (see 7380b0d, dead8ba). I also had to update the `miliseconds` in the `microuptime` implemenation as otherwise the calibration would turn into an endless loop. That being said, although there now is audio playback in VirtualBox, it sounds pretty much like it does in Qemu - but its a start.

nfeske, 2021-02-26:
The last activity on this issue was one year ago. @cnuke and @jjkarcher How should be go about it?

cnuke, 2021-02-26:
Running Sculpt with audio in a virtualized environment is still a non-existend use-case for me. I am fine with closing the issue and possible re-open it if we choose to work an that topic.

nfeske, 2021-02-26:
Thanks for your remark @cnuke.

closed 2021-02-26 by nfeske