genodelabs/genode #3222: fs_tool for performing file operations

nfeske, 2019-03-10:
We currently mostly use noux with coreutils for file operations like copying and removing files. This makes simple tasks like the prepare step of Sculpt more heavy weight than it needs to be.

In issue #3193 I suggested introducing a simple component called fs_tool for this purpose. Since the tool is only loosely related to the topic #3193, I'm taking it to this separate issue.

In contrast to my original suggestion, I think it would be good to define the operations in a more precise manner. I.e., instead of

```
<copy from="/path/to/origin" to="/path/to/destination"/>
```

It would be better to express what kind of thing to copy.

```
<copy-file from="/path/to/file" to_dir="/path/to/destination"/>
```

This way we can more easily extend the tool with more sophisticated operations in the future without stepping on our toes, e.g.,
```
<copy-files from_dir="/path/to/dir" wildcard="*.png" to_dir="/path/to/destination"/>
<copy-dir from="/path/to/dir" to_dir="/path/to/destination" as="another_name" recursive="yes"/>
```

In contrast to coreutils like `cp`, these explicit instructions leave no  room for ambiguities.

That said, in the first version, I will only implement the few simple operations, I need for #3193.

m-stein, 2019-03-10:
Sounds great. Thinking of this as long living server for all kinds of FS users made me think of the following scenario:
Imagine two arbitrary FS clients that like to execute an individual set of file operations on their FS but they don't know how and don't want to know. Instead they like to "pass" their FS session to the fs_tool instance temporarily together with an XML file containing the program to apply (written in the fs_tool configuration language). The fs_tool then would execute the program for the clients, inform them about the result and release access to the clients FS's.

Just a thought. To me this appears much more elegant than traditional methods like spawning and terminating an application for each operation. At the same time, the fs_tool would taint itself with access to the FS's only on demand.

nfeske, 2019-03-12:
The two commits above implement the initial version of the fs_tool component. At the moment, only the `<remove-file>' operation is supported (the only operation I need for #3193).

m-stein, 2019-03-14:
A fixup:
b39a8a8871 Fixup "gems: initial version of fs_tool" (improve runtime log event)

closed 2019-03-19 by nfeske