genodelabs/genode #3231: sculpt ce: sculpt_manager aborts if exception is thrown during deploy config update

ssumpf, 2019-03-14:
When there is a syntax error in config/deploy like `<xstart> .... </start>` a `Xml_node::Nonexistent_sub_node` exception is thrown and not catched by `Sculpt:Deploy::update_managed_deploy_config` which leads to `abort` being called by the sculpt_manager.

ssumpf, 2019-03-14:
@nfeske: 921bc94 wraps the XML parsing witch a `try/catch` which solves things for me. But I am not sure if this is the preferred solution.

nfeske, 2019-03-14:
Thanks for reporting. I think the problem lies somewhere else. The code in the sculpt manager should not raise any exception. If there is no sub node, the `for_each_sub_node` calls should do nothing. I suspect that the XML parser stumbles over the inconsistency of the node tags. I'll have a look.

commit ffc7eb84 (2019-03-14): base: never throw in Xml_node::for_each_sub_node

Fixes #3231
closed 2019-03-19 by chelmuth