genodelabs/genode #3232: nitpicker: dies with Lookup_failed exception during destroy_view

alex-ab, 2019-03-14:
A misbehaving client may kill nitpicker by invoking destroy_view() multiple times with same view handle.

```
[init -> test-label-bladex] Warning: good - destroy a handle
[init -> test-label-bladex] Warning: bad - bye bye nitpicker
[init -> test-label-blade] Warning: good - destroy a handle
[init -> nitpicker] Error: Uncaught exception of type 'Genode::Handle_registry<Genode::Handle<Nitpicker::View>, Nitpicker::View_component>::Lookup_failed'
[init -> nitpicker] Warning: abort called - thread: ep
Kernel: Segmentation fault (signum=11), see Linux kernel log for details
```

nfeske, 2019-03-15:
Thanks for reporting and the proposed fix. In https://github.com/genodelabs/genode/commit/059b2d77519c6f0b002991c0e4b39862f3cc43d3 I reworked the code such that the error case is handled correctly and the control flow becomes more obvious.

The patch also adopts your idea to add a bad-case test to testnit. Ultimately, I should really create a systematic automated test for nitpicker to catch regressions. But that's not for now.

alex-ab, 2019-03-18:
Just noticed, the wm has the same bug in the destroy_view method. Should I create a patch similar to yours for nitpicker or do you take care ?

nfeske, 2019-03-18:
I would appreciate that.

alex-ab, 2019-03-19:
I could not apply the same pattern due to the Locked_ptr semantic, nevertheless 07c1ff30fa5e77d54073d41de0ffc84af235db54 is my proposal to fix the issue for the wm.

nfeske, 2019-03-25:
Thanks for the fix @alex-ab. I merged https://github.com/genodelabs/genode/commit/07c1ff30fa5e77d54073d41de0ffc84af235db54 to staging.

alex-ab, 2019-04-02:
Fixed on master by c19c5ed0a4b1ca1e18d4c5a3b6f9fb03bf3f21e8

commit 059b2d77 (2019-03-15): nitpicker: fix destroy with invalid handle

This patch reworks the 'Session_component::destroy' to cope become
robust against a client-provided invalid view handle. The code did not
consider that 'Handle_registry::has_handle' may throw.

Thanks to Alexander Boettcher for reporting and the initial fix.

Fixes #3232
closed 2019-04-02 by alex-ab