genodelabs/genode #3235: prepare_ports: race condition with CONTRIB_DIR/.git

rite, 2019-03-18:
When preparing multiple ports in parallel, an error may occur during the initialization of the git repository in `CONTRIB_DIR`:

```
rm -rf contrib/.git
rm -rf $(./tool/ports/current acpica)
rm -rf $(./tool/ports/current bash)

./tool/ports/prepare_port acpica bash -j2

fatal: cannot copy '/usr/share/git-core/templates/description' to 'genode/contrib/.git/description': File exists
make[1]: *** [genode/tool/ports/mk/front_end.inc:58: genode/contrib/.git] Error 128
make: *** [tool/ports/prepare_port:33: bash] Error 2
make: *** Waiting for unfinished jobs....
```

This is because for every port to be prepared, the git repository in `CONTRIB_DIR` is initialized. This behavior has been introduced in a6acdac1cf3bcdf0f22b9f82a72a2fe6d3f127e8 (see #2959) and though it provides a reasonable safe guard, it breaks the parallel preparation of ports.

rite, 2019-03-18:
Although 49c8cbc fixes the problem by making the dependency explicit, its certainly not (yet) a valid solution as the duplicated definition of the variable `CONTRIB_DIR` is rather hacky...

Any suggestions how to proceed?

nfeske, 2019-03-21:
@emery since this issue is about the safety feature you introduced in https://github.com/genodelabs/genode/commit/a6acdac1cf3bcdf0f22b9f82a72a2fe6d3f127e8, could you please have a look?

ehmry, 2019-03-21:
@rite's patch looks fine to me if that solves the problem. I had hoped the safety Git repo would be initialized only once, perhaps the recipe should be for the file `$(CONTRIB_DIR)/.git/HEAD` rather than the directory `$(CONTRIB_DIR)/.git`?

rite, 2019-03-21:
@ehmry, according to the `git init` man page:
```
Running git init in an existing repository is safe. It will not overwrite
things that are already there. The primary reason for rerunning git init
is to pick up newly added templates (or to move the repository to another
place if --separate-git-dir is given).
```

I included your suggestion regarding `$(CONTRIB_DIR)/.git/HEAD` anyways as it increases the likelihood that we're indeed dealing with an actual Git repository and not just a random directory named `.git` - which Git could ignore and fall through...

I improved the fix in commit f713e09 by moving the definition of `CONTRIB_DIR` into `common.inc`.

nfeske, 2019-03-25:
Thanks @rite. I merged your commit to staging.

nfeske, 2019-04-02:
Fixed in master.

closed 2019-04-02 by nfeske